Ultimaker/Cura #4848: [3.x.x] Wall Support doesn't adjust to "Support Infill Layer Thickness"

TimoErdmann, 2018-11-21:
**Application Version**
3.x.x this Problem is present in the latest Version
**Platform**
Windows 10 64-bit
**Printer**
Custom FDM Printer (doesn't Matter)

**Steps to Reproduce**
1. add support, for example: "Zigzag" 
2. Change "Support Infill Layer Thickness" to 2x Layer Hight
3. Add 1 "Support Wall line Count" (or more doesn't matter)
4. slice

**Actual Results**
the outer support line (Wall) has the height of the default layer height (0.2 for example) and the support infill (zigzag-lines) have a height of double layer height (0.4 for example) 

This adds a lot of printing time

**Expected results**
Outer Support line (Wall) and inner Support (ZigZag) have the same layer Height

**Additional Information**
The turquoise lines are the support lines 
[image]
[image]

rburema, 2018-11-22:
Hi @TimoErdmann,
Note: The wall line count is the number of walls that get added in the horizonal direction. It doesn't add or compensate for height or thickness in the way you seem to expect.

With that in mind: As far as I can tell, the current outcome is the expected behaviour, since the setting you adjusted is for the (support) _infill_ layer thickness specifically.
cheers, Remco.

TimoErdmann, 2018-11-22:
Hi @rburema,
thank you for your reply. Yes the setting name lets except you that this will only change the "infill" support height. But there is no way to change the height of the support Walls. It is only possible to change the count of the walls. 

So is there a way to change the current behavior of the support wall height? Like change this with an additional setting? (For Example: "Outer Line Support Layer Height") Can you add this feature in the next Version of Cura?

rburema, 2018-11-23:
Hi @TimoErdmann,
I'm sorry, but due to the way other types of support work (concentric and tree for example), it seems that feature can't be added.
cheers, Remco.

closed 2019-02-22 by Ghostkeeper