Ultimaker/Cura #4849: [3.6] dropdown menu improvement request

Liger0, 2018-11-21:
**Application Version**
3.6

**Platform**
Win 10 x64

**Printer**
Cr-10s

**Steps to Reproduce**
Move when clicking or just use a non-professional mouse and click a dropdown, 

**Actual Results**
It clicks automatically on the already selected option even with a minimal movement.

**Expected results**
It should recignize the input being the one that opened the dropdown menu and not make it working as a dropdown selection too

**Additional Information**
[Video](https://www.youtube.com/watch?v=Z88a7KN14gY)

alekseisasin, 2018-11-21:
Do you have the same flickering with other dropdowns?

Liger0, 2018-11-21:
> 
> 
> Do you have the same flickering with other dropdowns?

I do.

alekseisasin, 2018-11-22:
So far I did not hear from other users something similar. On my Win10 and Ubuntu 18.04 works fine. What about the video card drivers?

Liger0, 2018-11-22:
> 
> 
> So far I did not hear from other users something similar. On my Win10 and Ubuntu 18.04 works fine. What about the video card drivers?

It is a cura fault caused by it not needing to release the mouse's left button in order to open the dropdown. Since the dropdown opens instantly when pressing the left button even if you don't release it, once it is opened, if you move on the right and go over a language and release the button it will select it and close the dropdown.

alekseisasin, 2018-11-23:
Hmm.. Sill cannot reproduce it, even on Mac. I think if you will try to run Cura on another computer everything will work fine.

Liger0, 2018-11-23:
> 
> 
> Hmm.. Sill cannot reproduce it, even on Mac. I think if you will try to run Cura on another computer everything will work fine.

I tried on 3 pcs with windows 10 and ubuntu, everyone has this fault.
It can't be avoided since it is how Cura manages the dropdown menus.
It should open the drodown when releasing the button, not when pressing it.

alekseisasin, 2018-11-27:
I got your proposal. 
What you are asking is not common behavior.  If you will open any application(web or desktop, simple example https://www.w3schools.com/tags/tryit.asp?filename=tryhtml_select) which has a dropdown you will see it opens after clicking and not after releasing button.

Liger0, 2018-11-27:
> 
> 
> I got your proposal.
> What you are asking is not common behavior. If you will open any application(web or desktop, simple example https://www.w3schools.com/tags/tryit.asp?filename=tryhtml_select) which has a dropdown you will see it opens after clicking and not after releasing button.

I tried that link and if I click on the "run" and other buttons, it works when releasing it, not while clicking.

If it has to work when clicking instead of when releasing, then it should know it has already clicked to open the dropdown and not click the language too in this particular case.

I can't see other options.

alekseisasin, 2018-11-29:
I was meaning the dropdown button "Volvo"(on the right section). When you click on it, the content of it appears. This is the same behavior what Cura has. In your video, Cura has the same dropdown button which shows the content after pressing the button and not when releasing.

GregValiant, 2026-09-10:
I'm going to close this as being stale.

closed 2026-09-10 by GregValiant