Ultimaker/Cura #4851: Prepare locks

Michaels-3dPrinting, 2018-11-21:
**Application Version**
3.6.0

**Platform**
Windows8

**Printer**
AM8

**Steps to Reproduce**
Changinging random parameters with Layer View and Prepare fails to work. Moving object like centre brings it back

**Actual Results**
As above

**Expected results**
As above

**Additional Information**
Otherwise everything seesms fine.  Seems stable after configuring printers that did not have configurations transfered from 3.4. Thank you - much better than 3.5 Keep up the good work

rburema, 2018-11-22:
Hi @Michaels-3dPrinting,
Do I understand correctly that; before moving a model to the center of the buildplate any paramters you set are ignored?
cheers, Remco.

Michaels-3dPrinting, 2018-11-22:
Thanks for the reply.

 

Version 3.6.0 Beta – Windows 8

 

Your understanding as indicated in your mail below - Not correct

 

If you have prepared item (got a slice model) and then change parameters on the right whilst having the view set to slice/layer it stops/ does not see change and stops the prepare button from working. When prepare button is locked - If you click on model and then ask to centre it unlocks the prepare button.

 

Is that any clearer?? Sorry if I am not explaining it explicitly.

 

Best regards

 

Michael

rburema, 2018-12-20:
We've reworked the interface in the mean time, perhaps you'd like to have a look at the [first beta](https://ultimaker.com/en/products/ultimaker-cura-software/download-request/209) and see if the problem still occurs.

rburema, 2018-12-29:
Hi @Michaels-3dPrinting,
Sounds like you might want to take a look at the 'Z Seam Alignment' setting. It does have a 'random' option, as well as some others.
cheers, Remco.

closed 2020-06-30 by Ghostkeeper