Ultimaker/Cura #4853: [3.6] the z increase is too low

Liger0, 2018-11-22:
**Application Version**
3.6

**Platform**
Win 10 x64

**Printer**
Cr-10s

**Steps to Reproduce**
I will link the gcode to try...

**Actual Results**
It is something extremely weird. It is the first time I use 0.2mm of layer height and the feature which makes a model different from the configuration (to use more infill than the others to be precise), I don't know if it means anything.
After doing the first layer, the z increase for the next one is so low that it is like it is redoing the layer again. It is the same for the zhop, even trying with 0.3 mm it just breaks through the print.
After some completely messed up layers, the extruder won't be able to push the filament because the nozzle is deeping inside the print, until the print is just removed from the buildplate. The first layer is good and it isn't too close to the bed, it is easy to remove... I tried to print it 5 times reslicing it.

**Expected results**
It should print normally

**Additional Information**

[gancio.zip](https://github.com/Ultimaker/Cura/files/2606354/gancio.zip)

smartavionics, 2018-11-22:
I've looked at your gcode and can't see any obvious problem, yet. 

Here it is starting the second layer at height 0.38, the first layer was at height 0.2. Looks OK.

[image]

smartavionics, 2018-11-22:
> Hi, what do you mean by this?

The gcode contains firmware retraction commands (G10 and G11). Some people use firmware retraction rather than using explicit G1 commands to retract and un-retract the filament.

Liger0, 2018-11-22:
> 
> 
> > Hi, what do you mean by this?
> 
> The gcode contains firmware retraction commands (G10 and G11). Some people use firmware retraction rather than using explicit G1 commands to retract and un-retract the filament.

I guess they are only for extrude at the start of the print to clean the nozzle and retract after the print finish?

smartavionics, 2018-11-22:
They are there because either the gcode flavour you are using enables firmware retraction or firmware retraction has been explicitly enabled (there is a setting to do that).

Liger0, 2018-11-22:
> 
> 
> I've looked at your gcode and can't see any obvious problem, yet.
> 
> Here it is starting the second layer at height 0.38, the first layer was at height 0.2. Looks OK.
> 
> [image]

Also, why 0.38mm of layer height? The layer height for both first and next layers should be 0.2, so the second one should be 0.40mm.

Liger0, 2018-11-22:
> 
> 
> They are there because either the gcode flavour you are using enables firmware retraction or firmware retraction has been explicitly enabled (there is a setting to do that).

I use marlin(volumetric) as a flavour, I don't know if it may cause it. I also can't see any firmware retraction setting.

smartavionics, 2018-11-22:
> Also, why 0.38mm of layer height? The layer height for both first and next layers should be 0.2, so the second one should be 0.40mm.

The first layer was 0.2, the second and subsequent layers are 0.18.

smartavionics, 2018-11-22:
The firmware retraction setting is hidden unless you have the printer settings plugin installed.

Liger0, 2018-11-22:
> 
> 
> > Also, why 0.38mm of layer height? The layer height for both first and next layers should be 0.2, so the second one should be 0.40mm.
> 
> The first layer was 0.2, the second and subsequent layers are 0.18.

Right, I used 0.18 as a test as it failed with 0.2mm before too, sorry.
Should I download the plugin from [here](https://community.ultimaker.com/topic/22743-printer-settings-plugin/)?

smartavionics, 2018-11-22:
The plugin is available using the marketplace...

[image]

Liger0, 2018-11-23:
> 
> 
> The firmware retraction setting is hidden unless you have the printer settings plugin installed.

Okay, I disabled the firmware retraction and now... I'll post a photo of the result.

[image]

As you can see the top wall of the taller model is completely skipped. There is not sign of overextrusion, so it didn't exen just extrude in the air.
There are 2 options: 
1. It moved in the air without extruding
2. It just stopped

Whatever is the reason, I didn't see any mechanical issue. The filament was flowing just fine extruding correctly via  the control box, so it isn't a machine extruding-related issue.
The cura preview shows the top layer though. But if even someone would think an extruding related issue (which as I said is not plausible), where wouldn't be any reason for it to do it exactly at the layer where the first top layer starts. It just ended precisely the latest infill layer and stopped extruding.

closed 2018-12-20 by Liger0