Ultimaker/Cura #4855: Temp changes on dual extruder setup - nozzle position

cmcgavigan, 2018-11-22:
Thank you for using Cura!
-->

**Application Version**
3.6.0

**Platform**
W10

**Printer**
Anet A8 with Geetech dual MK8 extruders

**Steps to Reproduce**
Normal Print

**Actual Results**
When printing pauses to change temps between extruders, the nozzle sits on the print. This pulls a string off the print when the nozzles move again, which gets embedded into the print.

**Expected results**
Carriage should move outside, or in middle of print tower to perform temp changes.

**Additional Information**
I need to do temp changes as my filament is finikity at low temps and oozy at high..

cmcgavigan, 2018-11-30:
Yes. When the printer wants to change to 2nd extruder, it must wait until 2nd hotend heats up first, whilst simultaneously cooling the first hotend by 10 degrees to prevent oozing. During this pause in printing, the nozzle that's cooling just stays at the last position it was extruding at, ie. on the print. It therefore sometimes burns, but always melts a small blob of printed material and pulls it off in a string when the heads move again. I note that other slicers do this temp change action on the prime tower away from the print. Sorry, this attach function will not let me attach a 3mf file for some reason..

rburema, 2018-12-11:
Thanks for your answer.

I wanted to try to reproduce this (read: see what this looks like in the gcode), but I see the Anet A8 isn't offered as one of the bundled printer-definitions. That means I'll really need to see either the 3MF file, or at the very least the printer-definition you're using before we'd be able to tell what's going on.

rburema, 2019-01-09:
Since I'm going through a few unresolved issues again:  How did you add the printer? Did you just add a custom FFF printer? Which settings where changed if any?

fughilli, 2019-05-28:
Hi all, I also have a modified Anet A8 printer. My setup has 3 nozzles, but for the purposes of resolving this bug, I can talk about my experience running only two nozzles (third disabled in Cura).

I have noticed that with my prints, the temperature change/nozzle switch happens without returning to the prime tower, and additionally, there is a very slow z-hop move during the nozzle temperature ramping which causes a small blob to form under the nozzle before switching. This blob would probably not be an issue on a larger print, but I am testing on a small model which very quickly becomes crufted with these blobs.

I am running Cura 4.1.0. I have attached my machine config, project file, and log.

[cura_bug_files.zip](https://github.com/Ultimaker/Cura/files/3229115/cura_bug_files.zip)

Meins321, 2019-09-18:
I have the same problems with my FlashForge Creator Pro (Dual Extruder)

i doesn't heat up/dump the new nose on the tower instead it snorts the wrong (new) material on the old print...

In cure a dual extruder needs to move off the model to the tower or wipeshield to clean the nose!

Ghostkeeper, 2019-09-19:
We have settings for this. They are called Extruder Start Position X, Extruder Start Position Y, Extruder End Position X and Extruder End Position Y. These have to be defined in the extruder definition of the printer in order for Cura to know where it should move at the end of an extruder plan and where it should come from when it starts a new extruder plan.

These settings can be either absolute coordinates or relative to the previous location (depending on Extruder Start Position Absolute and Extruder End Position Absolute). By default, it's set to be relative and coordinates 0,0 so indeed it will not move extra.

closed 2019-09-19 by Ghostkeeper