Ultimaker/Cura #4858: 3.6.0 mirror issues

Thisismydigitalself, 2018-11-22:
**Application Version**
3.6.0

**Platform**
Win10/64

**Printer**
CFFFP

**Steps to Reproduce**
Mirror an object

**Actual Results**
copy/mirrored object is not behaving the same as master

**Expected results**
Mirrored instance should slice the same

**Additional Information**
Base.3mf attached.
[image]
[image]

[BASE.zip](https://github.com/Ultimaker/Cura/files/2608435/BASE.zip)

Thisismydigitalself, 2018-11-22:
One more thing, Many months back i asked if it would be possible to have Cura show the user which object is a mirrored object. there is no way at least that i know of to know which is which.

Thisismydigitalself, 2018-11-22:
I believe I found the culprit - it was the "Make overhang printable". if checked, the mirrored part is broken.

frank26080115, 2019-01-13:
Happens in 4.0.0 BETA

[image]

Thisismydigitalself, 2019-01-13:
Same here
[image]

I stopped waiting for bugs to be fixed.
I opened an issue about this very same case almost a year and a half ago. case closed.
https://github.com/Ultimaker/Cura/issues/2917. 

I have a lot to say about 4.0 UX but since most of my comments and issues were ignored this passed two years or so, i keep quiet.

fieldOfView, 2019-01-14:
By running a version of Cura that is newer than the official beta

nallath, 2019-01-14:
I understand it's frustrating that the things you want / need don't get fixed, but there is no need to get all angry about it. There are simply too many things to do for us, so we simply have to perform triage on what to fix. 
In this case, the problem is with a very specific set of conditions (mirroring in combination with a extremely infrequently used setting), so it's just not as important as other issues. The same is unfortunately also true for showing what objects are mirrored; Not a whole lot of people use mirroring in the first place, so we can't defend putting a few days of time into making it work better.

Thisismydigitalself, 2019-02-10:
The most important thing to remember at all times is that we share the very same interest: to make Cura the best slicer out there. Having said that, I don't know who's working on what. i only feel like Cura failes to offer new and so basic features to support this tedious and sisyphic work of daily slicing and re-slicing over and over again and I don't see improvements in that sense.

benjjyman, 2020-05-16:
I'm just grateful cura us free I for one could not afford other named applications that charge large licence. I like the reprap movement, sharing knowledge so everyone gets a chance, not just the privelaged.
Thanks