Ultimaker/Cura #4860: Zits on curved walls

rlesac, 2018-11-22:
I have an issue with printing curved walls.
It manifests as zits on walls. 

The zits are always in the same position and most settings I tried have no effect on them.
Using other slicers (Slic3r, Simplfy3D) produces smooth walls.

The results are entirely reproducible between changing filament, nozzles, PTFE tube.

An extreme example:
[image]

Simple test STL:
https://www.thingiverse.com/thing:3229413

Simplify3D on top, Cura middle and bottom.  Setting z seam to user defined got rid of some zits in the sides of the oval, but others still persist.

24. Simplfy3D
27. z seam set to user defined
12. z seam set to shortest
[image]

[image]

Settings changed, from top to bottom:
18. combing off
16. reset to draft profile, extra prime amount to 0.4 (to counteract under extrusion after long travel, has no impact on the zits)
13. z seam alignment to sharpest
12. z seam alignment to shortest﻿
[image]

MF file
[CE3_Test retraction-extra prime amount v1.zip](https://github.com/Ultimaker/Cura/files/2608784/CE3_Test.retraction-extra.prime.amount.v1.zip)

**Application Version**
Cura 3.6

**Platform**
Win10 x64

**Printer**
Ender 3 stock.

**Steps to Reproduce**
Slice the file and print.

**Actual Results**
Zits and bumps on the walls

**Expected results**
Smooth walls

**Additional Information**

rlesac, 2018-11-22:
They even show up when printing the oval ring without the pins.
Always in the same place.
[image]
[image]

rlesac, 2018-11-25:
I also found new artifacts, vertical lines. Looking at the printer it seems to slow down in those places.
S3D also printed 40% faster.

S3d left and top. Cura right an bottom. Uh seems my photos got rotated upside down. Anyway, Cura is the one with all the zits.
[image]
[image]
[image]

Vertical lines:
[image]
[image]

Liger0, 2018-11-26:
Apart of the parts which need coasting and extra restart, I have the behaviour you have in your layers image on the right side too, that is very annoying.

Appesteijn, 2018-11-26:
Cura 3.6.0 & S5:

[image]

rlesac, 2018-11-27:
> Cura 3.6.0, Default 0.10 profile & S5:
> 
> [image]
So much brim :)  

I don't see any zits on your print.
Different layer heights produced different results with 0.2 being the worst and is my most often used layer height.

Do you have any ideas what could be causing such different behavior compared to other slicers?

I'll try some more test prints tomorrow.

Appesteijn, 2018-11-27:
So this is Cura 3.6 with 0.2 layerheight on a S5
[image]

rlesac, 2018-11-27:
> So this is Cura 3.6 with 0.2 layerheight on a S5
> [image]

This one seems to have the aforementioned zits on the walls, but it's hard to tell.
I have since switched from Marlin to Klipper firmware, and can't reproduce them with this test, I am getting different kind of artifacts, namely vertical lines.

But in other prints, more complex ones, they show up, as shown in the Star Trek cookie cutter.

[cc-star-trek-emblem-captain-70mm.zip](https://github.com/Ultimaker/Cura/files/2620730/cc-star-trek-emblem-captain-70mm.zip)

Only in Cura. Both Slic3r and Simplify3D are free of such defects.

I'm in the middle of a long print, so can¨do test at the moment.

Bobertso, 2018-11-28:
[image]
I have the same issue, found the cause by printing a lower resolution stl of my object. The zits are the model's vertexes, Cura's doing something weird at a vertex to make them. (low resolution on left, standard on right )

Solved by exporting model at extremely high resolution.

Liger0, 2018-12-20:
I even noticed weird behaviorus that show the inprecision of the skirt. Maybe this is related.

[image]

cshunt, 2018-12-21:
I think the zits are start/stops on layer change. I ran a couple test and if you set z seam to aligned I had them show up all in one row. I tried adjusting retract extra prime amount without success, this had helped in 3.4. Retract at layer change did not help. I thought it might be overlap at layer start/stop, coasting changed the shape of the zit.

Settings i have tried to adjust
Retract extra prime amount - worked in previous versions
retract on layer change - no change
z-seam location - this lined the zits up.
coasting

tablatronix, 2019-06-06:
What was the outcome of this, anyone resolve this, I seem to have this on certain layer heights and z seam settings

alfrix, 2019-06-09:
i have come across this yesterday printing a cylinder, i solved it by disabling "Compensate wall overlaps"

tablatronix, 2019-06-09:
Ill try that, I am printing lots of cylinders

nallath, 2020-11-26:
This is probably a combination of compensate wall overlaps and the resolution settings. A much more in depth writeup can be found on https://github.com/Ultimaker/Cura/issues/8321

closed 2020-11-26 by nallath

tablatronix, 2020-11-26:
Thanks, related ? #6676