Ultimaker/Cura #4862: [3.6.0] Y-axis starts 80° slope after a random Z height

yratof, 2018-11-22:
Supporting files:  [slicer problems.zip](https://github.com/Ultimaker/Cura/files/2609330/slicer.problems.zip)

**Application Version**
3.6.0

**Platform**
OSX 10.13.6

**Printer**
CR10s5

**Steps to Reproduce**
Simply slice and print via Octoprint, SD card or directly in Cura, all product exactly the same results

**Actual Results**
After 1cm of Z height, the Y axis starts to move the print layer by layer, perfectly towards the rear of the bed.

**Expected results**
It should print as shown in the Cura viewer and shouldn't start recreating the winter olympics ski ramp.

**Additional Information**
The Y slope happens at different points on different models.
To test that it's the gcode and not the machien, i sliced with 3DPrinterOS the exact same model and printed it, this model was flawless and printed without any problems.

smartavionics, 2018-11-22:
Hi. I have looked at cura.gcode in 2 gcode viewers and can see no obvious issues. I notice that the 3DprinterOS.gcode file doesn't contain any M204/M205 lines so if that prints OK, I would disable the accel and jerk in cura which will make the gcode smaller (and, who knows, maybe is related to the problem).

yratof, 2018-11-23:
I'll remove it, reprint the same gcode and report back

yratof, 2018-11-23:
Also worth noting that I’ve printed the same gcode 4 times and every time the shift and the slope start at exactly the same location

smartavionics, 2018-11-23:
Also worth noting that the gcode contains absolute x/y coordinates so it sounds like the Y axis is losing (or adding) steps. The question is why is it doing that? If you turn on z-hop does it still do it?

yratof, 2018-11-24:
Printed yesterday without Cura's acceleration or jerk settings and the default z hop and the print came out perfect, no flaws whatsoever. Not sure if it's a bug that clashes with TH3D unified firmware for the CR10s5, but without those settings, the print was not only faster, but was 100% amazing

yratof, 2018-11-24:
So not a default hardware issue — unless the acceleration is faster than the stepper can manage and causes step dropping

smartavionics, 2018-11-24:
Glad it's printing OK now. If you are happy, please close this issue. Cheers!

yratof, 2018-11-25:
Sure thing — hopefully this will help anyone in future issues

closed 2018-11-25 by yratof