Ultimaker/Cura #4866: Small support do not stick to bed

DavidThe1, 2018-11-23:
**Application Version**
3.6.0

**Platform**
windows 10

**Printer**
Ultimaker 2 and ultimaker 3 

**Steps to Reproduce**
 have small support surface less then 1cm x 1cm

**Actual Results**
the support don't stick to bed

**Expected results**
support unstuck from bed

**Additional Information**
 Would be great to have the possibility to have a full first layer for support touching the bed

nallath, 2018-11-27:
This is why there is the setting "Enable Support Brim"

closed 2018-11-27 by nallath

DavidThe1, 2018-11-27:
Ok thank you, but would be great to have this option for the support only.
most of the time we don't need brim for the part and it is time consuming
to trim it.
Best regards
David Gariépy

Integration Coach
+01 418-380-2788 <(418)%20380-2788> ext:210

Follow us on

[image: facbook-icone.png] <https://www.facebook.com/robotiq/> [image:
linkedin-icone.png] <https://www.linkedin.com/company/robotiq> [image:
twitter-icone.png] <https://twitter.com/Robotiq_Inc> [image:
google-icone.png]
<https://plus.google.com/u/0/b/100363269019698451101/+Robotiq> [image:
Youtube-icone.png]
<https://www.youtube.com/channel/UCHpdi06hmy0SSjrGqfmLZqQ>
Visit us:
<http://t.sidekickopen04.com/e1t/c/5/f18dQhb0S7lC8dDMPbW2n0x6l2B9nMJN7t5XYgdDSBWW1pNBMY7fRFTbW63BfNd56dW2Dd_ryyq02?t=http%3A%2F%2Frobotiq.com%2F&si=6062446723989504&pi=d6925f6b-b492-46d7-b57c-ed0842321b5e>
<http://t.sidekickopen04.com/e1t/c/5/f18dQhb0S7lC8dDMPbW2n0x6l2B9nMJN7t5XYgdDSBWW1pNBMY7fRFTbW63BfNd56dW2Dd_ryyq02?t=http%3A%2F%2Fdof.robotiq.com%2Fdiscussions&si=6062446723989504&pi=d6925f6b-b492-46d7-b57c-ed0842321b5e>

<http://t.sidekickopen04.com/e1t/c/5/f18dQhb0S7lC8dDMPbW2n0x6l2B9nMJN7t5XYgdDSBWW1pNBMY7fRFTbW63BfNd56dW2Dd_ryyq02?t=http%3A%2F%2Fblog.robotiq.com%2F&si=6062446723989504&pi=d6925f6b-b492-46d7-b57c-ed0842321b5e>

nallath, 2018-11-27:
That's what the setting does, it enables it for the support only.

DavidThe1, 2018-11-28:
Hi Jaime,

I didn't realized there was a specific setting that has to be enabled.
However, i enabled it and it still doesn't add any brim to the support.
[image: image.png]
[image: image.png]
David Gariépy

Integration Coach
+01 418-380-2788 <(418)%20380-2788> ext:210

Follow us on

[image: facbook-icone.png] <https://www.facebook.com/robotiq/> [image:
linkedin-icone.png] <https://www.linkedin.com/company/robotiq> [image:
twitter-icone.png] <https://twitter.com/Robotiq_Inc> [image:
google-icone.png]
<https://plus.google.com/u/0/b/100363269019698451101/+Robotiq> [image:
Youtube-icone.png]
<https://www.youtube.com/channel/UCHpdi06hmy0SSjrGqfmLZqQ>
Visit us:
<http://t.sidekickopen04.com/e1t/c/5/f18dQhb0S7lC8dDMPbW2n0x6l2B9nMJN7t5XYgdDSBWW1pNBMY7fRFTbW63BfNd56dW2Dd_ryyq02?t=http%3A%2F%2Frobotiq.com%2F&si=6062446723989504&pi=d6925f6b-b492-46d7-b57c-ed0842321b5e>
<http://t.sidekickopen04.com/e1t/c/5/f18dQhb0S7lC8dDMPbW2n0x6l2B9nMJN7t5XYgdDSBWW1pNBMY7fRFTbW63BfNd56dW2Dd_ryyq02?t=http%3A%2F%2Fdof.robotiq.com%2Fdiscussions&si=6062446723989504&pi=d6925f6b-b492-46d7-b57c-ed0842321b5e>

<http://t.sidekickopen04.com/e1t/c/5/f18dQhb0S7lC8dDMPbW2n0x6l2B9nMJN7t5XYgdDSBWW1pNBMY7fRFTbW63BfNd56dW2Dd_ryyq02?t=http%3A%2F%2Fblog.robotiq.com%2F&si=6062446723989504&pi=d6925f6b-b492-46d7-b57c-ed0842321b5e>