Ultimaker/Cura #4867: [3.6.0] Anycubic i3 mega doesnt shows material selection tab

dmolpas, 2018-11-23:
**Application Version**
Cura 3.6.0

**Platform**
win 10 64

**Printer**
Anycubic i3 mega

**Steps to Reproduce**
Just select the printer from the menu 

**Actual Results**
NO Material tab appears under the printer name . If you want to set the material proper temperatures you have to go by settings , profile , manage profile , material. 

**Expected results**
The Menu tab MATERIAL should appear under the printer name above print setup in the right part of the screen just as in any other printer 

**Additional Information**
Anycubic i3 mega
[image]
Anycubic 4 max (and many others)
[image]

edallas, 2018-11-23:
I can second that - on macOS Mojave with Cura 3.6.0 and Any Cubic i3 mega

shums, 2018-11-24:
It seems that Cura 3.6 supplied with wrong printer settings.
For WIN it can be overridden C:\Program Files\Ultimaker Cura 3.6\resources\definitions\anycubic_i3_mega.def.json 
(need administrator rights).

Change setting "has_materials" from **false** to **true**

edallas, 2018-11-24:
followed shums' lead - looked in application support but eventually found it in the application... /Applications/Ultimaker Cura/Contents/Resources/resources/definitions -> anycubic_i3_mega.def.json (you have to show package contents on the Cura.app to get there) 
Modified as suggested and reloaded Cura - now works on macOS Mojave!

dmolpas, 2018-11-24:
> It seems that Cura 3.6 supplied with wrong printer settings.
> For WIN it can be overridden C:\Program Files\Ultimaker Cura 3.6\resources\definitions\anycubic_i3_mega.def.json
> (need administrator rights).
> 
> Change setting "has_materials" from **false** to **true**

Thank you so much , works perfectly for me !

PR #5289 (2019-02-11): Fix Anycubic i3 Mega has_materials
"has_materials": true for Anycubic i3 Mega

Fixes #4867
commit 8e8bb930 (2019-02-11): Fix #4867
closed 2018-11-24 by dmolpas