Ultimaker/Cura #4869: Upgraded Cura, Printer Settings corrupt

Alphatango2, 2018-11-23:
**Application Version**
Cura 3.5.1 Upgraded to 3.6.0 (Issue Found) Downgraded back to 3.5.1 and Issue followed.

**Platform**
Windows 10

**Printer**
Custom FDM Printer

**Steps to Reproduce**
Upgrade

**Actual Results**
Corrupted Printer settings resulting in loss of my printer settings now printer does not print correctly before I upgraded.

**Additional Information**
[image]

nallath, 2018-11-26:
Could you share your logs and configuration data?

Alphatango2, 2018-11-26:
I will be out of town. When I return I will share the required info.

nallath, 2018-12-21:
I mean, can you share your entire configuration file? These are not all the files that caused the issue.

Alphatango2, 2018-12-21:
[Required InfosV2.zip](https://github.com/Ultimaker/Cura/files/2703241/Required.InfosV2.zip)
Here you are, the whole file.

nallath, 2019-01-07:
I've tried loading these into the upgrade system and those files seem to work just fine.

Alphatango2, 2019-01-08:
Hmm, well, I do not know what happened then, I guess ill just have to rework my settings, is there a way to look in the important files of cura to see if I can copy the settings I had out of those files so I dont have to go back through trial and error?

nallath, 2019-01-08:
If you use the reset option, it will make a zip file with the corrupted settings in it as a sort of a backup. If you can send me those files, i could have a look.

Alphatango2, 2019-01-08:
Where do those files go when I hit the reset, in the root folder of Cura or elsewhere?

fieldOfView, 2019-01-08:
Help -> Show configuration folder. Then go one folder up.

Alphatango2, 2019-01-09:
[3.5_20181123_173113.zip](https://github.com/Ultimaker/Cura/files/2738846/3.5_20181123_173113.zip)
[3.6_20181123_104114.zip](https://github.com/Ultimaker/Cura/files/2738847/3.6_20181123_104114.zip)
[3.6_20190108_155546.zip](https://github.com/Ultimaker/Cura/files/2738848/3.6_20190108_155546.zip)

Alphatango2, 2019-01-09:
Sorry about that, so the 3.5 zip is when i downgraded, and the 2018 3.6 is when I would have upgraded. and the 2019 3.6 is from today.

nallath, 2019-01-09:
Well, it's now clear why it's failing, for some reason the user profiles are not stored at all (the changes that you didn't save into a profile yet). Why this occurred is a total mystery to me. 

The user profiles are available in the 3.5 folder though. What you can do is restore the 3.5 folder and delete the 3.6 one. That should trigger the upgrade script again.

Alphatango2, 2019-01-09:
Well, at least we have a clear idea as to why it was failing. Thank you for your patience and help. So just to clarify, I currently have 3.6 installed, do I uninstall and reinstall 3.5 unzip the zip folder and add my own 3.5 file to the 3.5 folder?

nallath, 2019-01-10:
Yup. If you remove the 3.6 folder and you have a 3.5 folder (and the cura you start is 3.6) cura will copy the files it finds in the 3.5 folder to the 3.6 folder and subsequently try to upgrade it.

Alphatango2, 2019-01-11:
Update : I still receive the corrupted file issues that are displayed when Cura starts up but I looked over my setting and they all show what I had them set to. I tried a gcode on my printer and it printed just fine. (bed needs to be releveled) but beyond that I am able to save my gcode file with my setting even with that display message,

nallath, 2019-01-14:
You could try and remove the files that it mentions in the warning message. It's just that those files can't be loaded (for some reason)

si-hb, 2019-04-25:
Cura 4.0 has been doing the same thing to me. Asking me to reset my settings every time I launch it, which I had been refusing, but now I have done so because it was persistent and I have then lost all of my custom materials settings. Have to re-input all my material costs and re create custom materials.

closed 2020-06-20 by no-response[bot]