Ultimaker/Cura #4870: [3.6.0] "Cura can't start"

mpsdskd, 2018-11-23:
**Application Version**
3.6.0

**Platform**
Kubuntu 18.10

**Printer**
Custom FDM

**Steps to Reproduce**
Start Program

**Actual Results**
Error Window, Program does not start

**Additional Information**
I tried the "backup and restart" option in the error window - I then can open Cura 3.6 but cannot restore my config using the backup addon.

[cura.log](https://github.com/Ultimaker/Cura/files/2611517/cura.log)
[3.6_20181123_194345.zip](https://github.com/Ultimaker/Cura/files/2611518/3.6_20181123_194345.zip)

"Error Traceback" in Error Window
> Traceback (most recent call last):
>   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
>   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
>   File "/build/env/3.6/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12, in <module>
>   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
>   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
>   File "/build/env/3.6/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21, in <module>
>   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
>   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
>   File "/build/3.6/build/inst/bin/cura", line 138, in <module>
>   File "/build/3.6/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 686, in run
>   File "/build/3.6/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 287, in run
>   File "/build/3.6/build/inst/lib/python3.5/site-packages/UM/Application.py", line 207, in run
>   File "/build/3.6/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 426, in startSplashWindowPhase
>   File "/build/3.6/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 185, in startSplashWindowPhase
>   File "/build/3.6/build/inst/lib/python3.5/site-packages/UM/VersionUpgradeManager.py", line 137, in upgrade
>   File "/build/3.6/build/inst/lib/python3.5/site-packages/UM/VersionUpgradeManager.py", line 341, in _upgradeFile
>   File "/build/3.6/build/inst/lib/python3.5/site-packages/UM/VersionUpgradeManager.py", line 398, in updateFilesData
> TypeError: 'NoneType' object is not iterable

"Error Log" in Error window

> Thread 0x00007f19137fe700 (most recent call first):
>   File "/build/3.6/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 160 in _storeStderrToLogThread
>   File "/build/env/3.6/inst/lib/python3.5/threading.py", line 862 in run
>   File "/build/env/3.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
>   File "/build/env/3.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap
> 
> Thread 0x00007f1913fff700 (most recent call first):
>   File "/build/3.6/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 152 in _storeOutputToLogThread
>   File "/build/env/3.6/inst/lib/python3.5/threading.py", line 862 in run
>   File "/build/env/3.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
>   File "/build/env/3.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap
> 
> Thread 0x00007f19299ea700 (most recent call first):
>   File "/tmp/.mount_fsLn1y/usr/bin/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 78 in _updateThread
>   File "/build/env/3.6/inst/lib/python3.5/threading.py", line 862 in run
>   File "/build/env/3.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
>   File "/build/env/3.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap
> 
> Thread 0x00007f192a7fc700 (most recent call first):
>   File "/build/env/3.6/inst/lib/python3.5/threading.py", line 297 in wait
[... 100 lines cut ...]
> 2018-11-23 19:53:19,284 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
> 2018-11-23 19:53:19,290 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/urllib/request.py", line 1254, in do_open
> 2018-11-23 19:53:19,297 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/http/client.py", line 1106, in request
> 2018-11-23 19:53:19,303 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/http/client.py", line 1151, in _send_request
> 2018-11-23 19:53:19,310 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/http/client.py", line 1102, in endheaders
> 2018-11-23 19:53:19,317 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/http/client.py", line 934, in _send_output
> 2018-11-23 19:53:19,323 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/http/client.py", line 877, in send
> 2018-11-23 19:53:19,330 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/http/client.py", line 1260, in connect
> 2018-11-23 19:53:19,337 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/ssl.py", line 377, in wrap_socket
> 2018-11-23 19:53:19,343 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/ssl.py", line 752, in __init__
> 2018-11-23 19:53:19,350 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/ssl.py", line 988, in do_handshake
> 2018-11-23 19:53:19,357 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/ssl.py", line 633, in do_handshake
> 2018-11-23 19:53:19,363 - ERROR - [MainThread] UM.Logger.logException [88]: ssl.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:645)
> 2018-11-23 19:53:19,370 - ERROR - [MainThread] UM.Logger.logException [88]: 
> 2018-11-23 19:53:19,376 - ERROR - [MainThread] UM.Logger.logException [88]: During handling of the above exception, another exception occurred:
> 2018-11-23 19:53:19,383 - ERROR - [MainThread] UM.Logger.logException [88]: 
> 2018-11-23 19:53:19,390 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
> 2018-11-23 19:53:19,396 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/3.6/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 369, in _sendCrashReport
> 2018-11-23 19:53:19,403 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/urllib/request.py", line 163, in urlopen
> 2018-11-23 19:53:19,410 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/urllib/request.py", line 466, in open
> 2018-11-23 19:53:19,416 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/urllib/request.py", line 484, in _open
> 2018-11-23 19:53:19,423 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/urllib/request.py", line 444, in _call_chain
> 2018-11-23 19:53:19,429 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/urllib/request.py", line 1297, in https_open
> 2018-11-23 19:53:19,436 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/build/env/3.6/inst/lib/python3.5/urllib/request.py", line 1256, in do_open
> 2018-11-23 19:53:19,443 - ERROR - [MainThread] UM.Logger.logException [88]: urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:645)>

Appesteijn, 2018-11-23:
Do have have a zip file of you settings folder? We are looking to reproduce this error for some time now but couldn't.

mpsdskd, 2018-11-23:
It is attached in the link below the link to the log in the original post.
[](https://github.com/Ultimaker/Cura/files/2611518/3.6_20181123_194345.zip)

Ghostkeeper, 2018-11-26:
Fixed in the next release here: https://github.com/Ultimaker/Cura/commit/ae2b3124721c9c74730073d178c0ac1927819b1c

Also fixed an underlying issue here that would've prevented the buggy code from being called in the first place: https://github.com/Ultimaker/Uranium/commit/57a16e55c93ce0de5f392209144ca46aeba10862

Thanks for the report! We'd seen this bug before via the crash window's report button but your settings folder was helpful to reproduce the bug ourselves and then trace it down to verify that we had actually fixed it.

closed 2018-11-26 by Ghostkeeper

mpsdskd, 2018-11-26:
I'm glad I could help.
Thank you very much for (hopefully) solving the problem :+1: