Ultimaker/Cura #4871: [3.6.0] Inefficient small gap filling between walls

lekv, 2018-11-23:
**Application Version**
3.6.0

**Platform**
MacOS Mojave

**Printer**
Creality CR-10 with defaults (the printer doesn't seem to matter, it happened with an Ender 3 Pro, too.

**Steps to Reproduce**
Slice a part that has a 2mm thick solid wall using a 0.4mm nozzle and 0.4mm wall thickness. Set wall-count to 2.

**Actual Results**
The whole part is 5 walls thick. That leaves 0.4mm in the middle, which should be filled with a single extruded line. In 3.5.1 this works as expected, but in 3.6.0 the gap is filled with a diagonal pattern instead, which leads to longer print times (and potentially less strength in the resulting part).

Additionally, parts that are 1.2mm / 3 lines wide get the same infill instead of a proper wall. If I change the line width to 0.399mm, then those get a single inner wall line inside, but the inside of a 5 line wide wall will still stay the same. Reducing the line width to 0.39mm causes the inside of the 5 line wide wall to become empty / standard infill pattern, like it wasn't considered a "gap" to be filled anymore.

Turning off "Fill Gaps Between Walls" leaves the whole space unfilled in either case.

**Expected results**
See above, like in 3.5.1 the gap should be filled with a single extrusion.

**Additional Information**

This is how it looks in 3.6.0. Note how the part on the left hand side and the left wall of the lower two have a diagonal filling.

[image]

This is how it looks with the width set to 3.99mm. Both lower walls now have a single line inside, but the left wall still doesn't.
[image]

This is how it looks in 3.5.1 (3.4.1 too), which makes me believe that this is a regression.
[image]

smartavionics, 2018-11-24:
It is a feature of my implementation that some gap fill lines are only drawn every other layer. That is normal!

[image]

smartavionics, 2018-11-25:
Can you build Cura now using cura-build-environment and cura-build repos? Once you can do that it's not difficult to switch to using my repos for the parts that I have changed.

I don't know why it has regressed but, then again, I do not track what the Cura devs do other than to react to the bugs when I find them!

smartavionics, 2018-11-25:
I am now using my replacement gap filling implementation to also fill the outline gaps that occur when printing thin walls. Needs more testing but it seems to work OK. My next release (later today) will contain that also.

JohnEdwa, 2018-12-02:
I think this is related/the same as [the issue I posted around a month back.](https://github.com/Ultimaker/Cura/issues/4685)

smartavionics, 2018-12-02:
Hi @JohnEdwa , yes you're right it's the same problem.

JohnEdwa, 2018-12-03:
@smartavionics I tested your master build and can confirm that it fixes issue #4685.    
 I have around 15 copies of a print I need to do for christmas and this fix dropped the time in half, so I'll keep using and testing it, thank you!

smartavionics, 2018-12-03:
I'm glad it's been useful for you - my gap filling and thin line code appears to produce nice quality output but I think it can be slow to slice if the model has a lot of gaps and/or thin walls. For me, print quality and time is far more important than slicing time. Cheers!

BagelOrb, 2018-12-04:
@smartavionics which PR are you referring to?

smartavionics, 2018-12-04:
BTW, I have completely disabled the call to optimizePaths() as I believe it is the root cause of lots of evil. It seems to break various infills, ironed skin and also can't do a decent job of converting the zig-zag gap fill and thin wall regions into plausible single extrusions. To be honest, I think that approach (printing the gaps as zig-zag infill and then hoping that the paths can be optimized into variable width single extrusions) is not the way to go which is why I have come up with an alternative implementation.

BagelOrb, 2018-12-04:
I was planning on solving it in a very principled and complicated manner
over the next year or so.

I'm curious: what's your approach?

BagelOrb, 2018-12-04:
> Hi Tim, the inter-wall gap filling and thin wall printing come down to the
> same basic problem, filling a thin polygon. I use the same code for both
> jobs. I show the entire code below.
>
> There's basically two parts to the code, the first part is straightforward
> and that simply looks at the polygons and from the vertices generates a new
> set of points that approximate the centreline of the gap. Those are held in
> the vector mid_points. I also generate a rectangular polygon that
> approximates the area that would be filled for each line segment. The
> corner points of that rectangle are stored in begin_points and end_points.
>
> The second part of the code iterates through mid_points and for each line
> segment prints out a line of suitable width. The hard part here is to track
> which line segments have been printed so that subsequent line segments can
> be omitted if they would overlap with the already printed line segments.
>
> It's not hard conceptually but it's hard to get an efficient
> implementation. I tried using a sparse point array to take advantage of
> locality but the code was more complex and no more efficient than the
> current implementation which simply keeps track of which line segments have
> been printed by unioning their area rectangles together to form a polygon
> which can then be intersected with the area rectangle for each line segment
> that may yet be printed.
>
> I will continue to fiddle with it to see if I can up the efficiency but as
> [... 158 lines cut ...]
>                 next_poly_index = cpp.poly_idx;
>             }
>         }
>     }
> }

Sounds like your idea is definitely more accurate than the current
implementation.

Perhaps you should make a PR and discuss the implementation and improvement
ideas with the Cura team.

TimoErdmann, 2018-12-06:
This is brilliant. This slicer benefits extreme with your implementation of filling small gaps! I appreciate your work.

smartavionics, 2018-12-06:
Thanks @TimoErdmann. It's good to know that it is working well for you.

BagelOrb, 2018-12-06:
Here's an extreme stress test for thin walls and gaps: 
[thin_walls_stress_test.svg_1mm.stl.zip](https://github.com/Ultimaker/Cura/files/2652804/thin_walls_stress_test.svg_1mm.stl.zip)

Try different line widths to test this model.

smartavionics, 2018-12-06:
Thanks for that @BagelOrb. It doesn't handle that very well right now, I will see if I can improve it.

[image]

smartavionics, 2018-12-06:
Getting better...

[image]

JohnEdwa, 2018-12-11:
@smartavionics Just a heads up, in Master 20181210 the "Minimum Infill Area" function has broken again Not sure if it was on earlier builds as well and might not have anything to do with your changes but I don't think creating new issues for these builds in this repo is really appropriate either.

Values above 0 but below the size it would activate don't generate bottom layers.
[Minimum Infill Area: 0](https://user-images.githubusercontent.com/5417183/49823381-34527980-fd88-11e8-8dd8-3e232093d87d.PNG)
[Minimum Infill Area: 50](https://user-images.githubusercontent.com/5417183/49823383-34527980-fd88-11e8-95ad-c4377665641a.PNG)
[Minimum Infill Area: 750](https://user-images.githubusercontent.com/5417183/49823379-34527980-fd88-11e8-8319-34b0651b1a1f.PNG)

smartavionics, 2018-12-11:
Hi @JohnEdwa , could you please share that project so I can have a play. Thanks.

JohnEdwa, 2018-12-11:
@smartavionics Sure, it's a slightly modified one as I added a sacrificial bridge and don't have the old file, but it still clearly shows the issue on layer 51 when you change Minimum Infill Area.

[CE3_Component4.zip](https://github.com/Ultimaker/Cura/files/2669263/CE3_Component4.zip)

smartavionics, 2018-12-11:
Thanks for the project. I had a play with it and it does seem screwy. I think I was the original creator of that setting but it subsequently got reworked by the Cura team. I don't actually use it anymore because I tend to use the skin shrink and expand settings instead.

smartavionics, 2018-12-11:
The min infill area feature does appear to act weird - it is affected by the use of skin removal width. Your profile has a value of 0.8mm for that. If you reduce it to zero then that central region of skin on layer 51 doesn't look too bad with a min infill area value of 30 but does go wrong when the min infill area is increased to 50. To be honest, I can't really be bothered to find out what they have broken exactly, life's too short!

Diederikvanderpant, 2019-01-02:
@smartavionics ,
thanks for this feature. I finally managed to print this led armature in clear PET without ugly scars from unnecesary movcements.  Had to look hard for this option though. So for others who find this thread I will share the breadcrumbtrail here.

I added options in section: "_shell_" --> "_compensate wall overlaps_" and "_minimum wall flow_" 
I checked _compensate wall overlaps_ and set _minimum wall flow_ to 15% .

[image]

Diederikvanderpant, 2019-01-04:
@smartavionics 

@BagelOrb was curious of the product. And today I got the fadecandy ledcontroller and the rpi working. So I made some nice pics. Credits for the nice glowing effect go to the creator of fadecandy.
https://github.com/scanlime/fadecandy.

[image]
[image]
[image]

BagelOrb, 2019-01-05:
Cool!

Thanks!

torsoreaper, 2019-05-01:
[image]
[image]

Just adding some context since it appears this issue is still causing some problems.  The pic with the infill inside the thin wall is Cura and the other is from Slic3r.  With Slic3r I can almost set wall count or width to anything except 1 and it will fill the thin wall with a line.  In Cura no matter what I did, I could not get it to fill with a line.

Ghostkeeper, 2022-04-10:
Fixed in master, to be released with Cura 5.0. The gap filling has been replaced with variable line width walls.

closed 2022-04-10 by Ghostkeeper