Ultimaker/Cura #4873: Change number of walls/perimeters for first 4 layers

debil1, 2018-11-24:
I tried all scripts but can't found a solution to change from layer 0 to 4 -> 2 walls/perimeter and then further on only one.

nallath, 2018-11-26:
You could try and do this with infill meshes. Put an infill mesh on the first few layers.

diegopradogesto, 2018-11-28:
Yes, so you can place another mesh at the bottom and use the per object settings (e.g. "Modify settings for overlap with other models") and then select the settings you want to change, such as `Wall Line Count` and `Infill Density`.

Take a look at the picture:

[image]

BagelOrb, 2018-12-07:
With zero percent infill that can be a problem. Just put infilll to 0.1% and set the infill offset X and y to large values so that there is factually no infill in your part.

closed 2020-06-19 by nallath