Ultimaker/Cura #4874: No retraction at all

Wonderhamster, 2018-11-24:
I've been using Cura since a couple of months now it's worked fine. After I downloaded the last update(3.6.0) retraction settings stopped having any effect on the print. I checked every settings, thar is related to it, but nothing changed. I started to realize, the problem is with the new update, so I exported the printing profile, and imported it to Cura 3.5.1, and tried to generate the same G-code, and the result is indeed that the new Cura was the problem. 

[Files.zip](https://github.com/Ultimaker/Cura/files/2612192/Files.zip)

**Application Version**
 3.6.0 

**Platform**
Windows 10

**Printer**
Tevo Tarantula

smartavionics, 2018-11-24:
The 3.5.1 gcode is using explicit G1 E? lines to retract and un-retract. The 3.6.0 gcode is using G10/G11 firmware retraction. Perhaps your printer is not setup to do firmware retraction?

sozkani, 2018-11-27:
Perhaps your printer is not setup to do firmware retraction! Because we are not dealing details. Just simple user, check for use retract or not!. Do you plan to revert back or any choice for new firmware or old firmware? 
Thanks.

Budge72, 2019-01-03:
Hi.  I'm having the same problem.  I checked and my Marlin firmware is has M207 commented out for firmware retraction.  This is definitely a bug in Cura, though.  Because why have retraction settings in Cura just to have it produce a gcode with G10/G11 commands that will ignore those settings?  And why would I want to use firmware retraction settings?  I have different retraction settings for different filaments.  This would be a lot simpler if there was a toggle in the Cura printer profile that allowed us to choose Cura or firmware retraction settings.

smartavionics, 2019-01-03:
There is a setting called Firmware Retraction, it defaults to true (on) when using Marlin (volumetric) which is what @Wonderhamster had. It does not appear in the UI unless you have the Printer Settings plugin. Arguably, it is a bug that you cannot easily change its value without that plugin or editing the printer definition file.

smartavionics, 2019-01-04:
It should be available in the toolbox (aka marketplace).

kaovilai, 2019-04-04:
bump. Having same issue on Ender 3 stock firmware through Octoprint

kaovilai, 2019-04-04:
also resolved through the plugin

Ghostkeeper, 2019-04-05:
Tevo Tarantula has the default g-code flavour set to "RepRap (Marlin/Sprinter)" though, not "Marlin (Volumetric)".

MisterPrintf, 2019-04-20:
Try installing the printer settings plugin and disabling machine retraction. That solved it for me.

Wonderhamster, 2019-04-21:
Few months passed and I downloaded Cura 3.6 again. No problem this time, everything works like a charm. Thank you all for your help.

closed 2019-04-21 by Wonderhamster