Ultimaker/Cura #4876: Anycubic i3 Mega Quality Profiles not loaded Cura 3.6.0

hanturk, 2018-11-24:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

hanturk, 2018-11-24:
Because entry "global_quality = True" is missing at the three Quality Profiles under "C:\Program Files\Ultimaker Cura 3.6\resources\quality\anycubic_i3_mega" they are not loaded to Cura 3.6.0. After adding this value to the profiles there are loaded like in Cura 3.4.1. I'am not shure if this correction is O.K. the author of the profiles @thetobby should check this issue.

thetobby, 2018-11-24:
Feel free to take over or submit improvements. I'm not activly maintaining the profiles so might be bugs with never versions.

diegopradogesto, 2018-11-28:
I just did it. Thanks for reporting!

closed 2018-11-28 by diegopradogesto

thetobby, 2018-11-28:
Thank you @diegopradogesto

hanturk, 2018-11-28:
Thank you all. Have a nice day.

commit 709750c9 (2018-11-28): Add global quality to the Anycubic i3 Mega

Fixes #4876.