Ultimaker/Cura #4877: [Suggestion]  Toggle between different slice data without re-slicing again and again

Thisismydigitalself, 2018-11-24:
Cura should allow us users to toggle between differnet slices generated by differnet profiles. More and more i find myself slicing again and again with differnet settings and wished I could compare between those without the need to re-initiate slicing. much like generative design where CAD offers the designer to chose between differnet versions for best outcome. wouldn't it be nice to be able to chose few profiles and ask Cura to slice those and keep the sliced results, data and rendered view, for us to toggle through and chose the best one for print?

nallath, 2018-11-26:
I totally get why you want this, but a feature like that is going to be a massive memory hog.

Thisismydigitalself, 2018-11-26:
Hi, Cura with nothing loaded consume about 550Mb of ram. loaded and sliced complex model which takes about 19h print time (gcode larger than 50Mb) and Cura is up to 650Mb in ram. if i load ~5 iterations of a much smaller model than the one i tested - don't you think this is doable? it seems doable but then again, i might be missing something here.

BagelOrb, 2018-12-07:
Maybe your workflow could simply consist of opening Cura twice and slicing a file on both curas with a different profile.

Thisismydigitalself, 2018-12-07:
@BagelOrb
Been there, done that, but that defeats the purpose. 
I was hoping to toggle between sliced options from single instance of Cura or even better to see like 4 or 6 in one render screen, but i understand my wish is hard to comply.  just giving some ideas of how workflow can be helpful.

Awaits Cura 5.0. 
I'll act surprised :)

ozfunghi, 2018-12-17:
> 
> 
> Maybe your workflow could simply consist of opening Cura twice and slicing a file on both curas with a different profile.

Hi BagelOrb... a bit off topic, but still connected: is it necessary for the slicer to slice again each time settings are changed, that do not affect the actual slicing. Let's say i increase the bed temperature, the entire slice result needs to be calculated again, while the bed or nozzle temp do  not affect the sliced layers AFAIK. The same thing for changing the fan speed, or other stuff like that.

BagelOrb, 2018-12-17:
The bed temperature does have effect because together with the initial
layer bed temp and the warm up speed the position in the gcode is chosen
wehre to put the command to start heating the bed.

Same holds for fan speed. The fan speeds change depending on layer geometry
and size and the actual fan speed commands are determined by a lot of
settings.

This all goes to show that the feature you are asking for is a lot more
difficult and there is always a catch.

I don't think the feature you suggest will ever be implemented and I think
it's better for everybodys sanity if we don't.

Thisismydigitalself, 2018-12-18:
Slicing a 3D model with different parameters has profound implications on part properties. i would bet it's just a matter of time before this feature will be implimented. in the near future we will tell our slicers which material we're printing with, how much time we have, what's the expected weight/mass  & strength required and the slicer will generate few options to choose from - Much like CAD's generative design.