Ultimaker/Cura #4887: Ultimaker cura version 3.6 takes too long when loading

litolifestyle, 2018-11-27:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

alekseisasin, 2018-11-27:
This is very strange. If you will open Cura 3.0 you will immediately the difference.
Maybe you can share the log file.
Or you can try to clean Cura settings in the system folder.

Note: Please fill the issue form correctly, otherwise Cura team cannot help you. It required time to ask extra questions, (like OS type, printer type, reproduce steps and etc.)

Thisismydigitalself, 2018-11-27:
I concur. 3.6 on win10/64 loads between 16-19sec which is almost double the time Cursa 3.4.1 or even earlier veriosn were at around 9 seconds.

Liger0, 2018-11-28:
I have always had a very long loading time no matter what version or using an SSD.

Christian72D, 2018-11-28:
Same here, on my laptop Cura 3.6 takes 3-4 times longer to start compaed to 3.4.1 or 3.5.1.

Liger0, 2018-11-30:
> 
> 
> What if you will try to "Run as administrator" Cura.
> Try to start Cura without an internet connection,

So, for the first time I start it up after the PC has been booted, it needs about a minute, while closing and reopening it it needs at least 20 seconds.

Thisismydigitalself, 2018-12-02:
>  Try to start Cura without an internet connection,

Given "Check for updates" and "send print information" are both unchecked, What does Cura communicate during startup?

closed 2020-06-20 by no-response[bot]

Ghostkeeper, 2020-06-30:
> Given "Check for updates" and "send print information" are both unchecked, What does Cura communicate during startup?

If you were logged in when closing Cura last time, it'll contact the authentication servers to log you back in. Cura will also announce in your LAN to ask whether there are any printers there. I think Alekseisasin was thinking in this direction because the Ultimaker office network has hundreds of printers and processing the responses from all of those can take some time.