Ultimaker/Cura #4890: [Suggestion] - Sync Layerdata with Octoprint current printing layer

Thisismydigitalself, 2018-11-27:
Split Monitor & Prepare windows Side by Side and listen to OctoPrint current layer being printed.
That can be cool practical and helpul to see in real time and in 3D what comes next and to compare the outcome with what was requested so the user can intervene in real time and change parameters during print.

fieldOfView, 2018-11-27:
It is a cool idea, but may not be practical; the layer data may no longer exist in Cura. If you change a setting in the prepare screen, the layer data is gone from Cura even if it is still printing in OctoPrint. You can even load and slice an entirely different model in Cura while printing in OctoPrint.

diegopradogesto, 2018-11-28:
This is something similar as #3509. Quite hard I guess.

nallath, 2018-11-28:
I feel this is more of a feature for OctoPrint itself (eg; Octoprint has all the data in this case. It should know what it's printing and where it is). As @fieldOfView already mentioned, it's going to be quite hard to get this to work in Cura.

Thisismydigitalself, 2018-11-28:
Ok, thank you all for at least considering this.

closed 2023-07-13 by github-actions[bot]