Ultimaker/Cura #4894: Cura 3.6 Crashes on load

akf-was-here, 2018-11-27:
**Application Version**

3.6.0

**Platform**

Windows 10 64bit

**Printer**

None, won't load at all

**Steps to Reproduce**

Start the program.  Looks like it's bombing out at "Loading Interfaces".

**Actual Results**

Windows message "Cura.exe has stopped working" 

**Expected results**

Program shouldn't crash.

**Additional Information**

cura.log:

2018-11-27 09:32:55,601 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FileLogger
2018-11-27 09:32:55,616 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FirmwareUpdateChecker
2018-11-27 09:32:55,631 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FirmwareUpdater
2018-11-27 09:32:55,645 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeGzReader
2018-11-27 09:32:55,658 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeGzWriter
2018-11-27 09:32:55,672 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeProfileReader
2018-11-27 09:32:55,692 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeReader
2018-11-27 09:32:55,708 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeWriter
2018-11-27 09:32:55,724 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ImageReader
2018-11-27 09:32:55,739 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LegacyProfileReader
2018-11-27 09:32:55,752 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalContainerProvider
2018-11-27 09:32:55,772 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalFileOutputDevice
2018-11-27 09:32:55,786 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MachineSettingsAction
2018-11-27 09:32:55,801 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ModelChecker
2018-11-27 09:32:55,818 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MonitorStage
2018-11-27 09:32:55,834 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PerObjectSettingsTool
2018-11-27 09:32:55,851 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PostProcessingPlugin
2018-11-27 09:32:55,868 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PrepareStage
2018-11-27 09:32:55,888 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin RemovableDriveOutputDevice
2018-11-27 09:32:55,908 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SimulationView
2018-11-27 09:32:55,927 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SliceInfoPlugin
2018-11-27 09:32:55,945 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SolidView
2018-11-27 09:32:55,965 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SupportEraser
2018-11-27 09:32:55,988 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin Toolbox
2018-11-27 09:32:56,009 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CameraTool
2018-11-27 09:32:56,030 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MirrorTool
2018-11-27 09:32:56,052 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin RotateTool
2018-11-27 09:32:56,072 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ScaleTool
2018-11-27 09:32:56,093 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SelectionTool
2018-11-27 09:32:56,114 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin TranslateTool
2018-11-27 09:32:56,146 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UFPWriter
2018-11-27 09:32:56,168 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UltimakerMachineActions
2018-11-27 09:32:56,227 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UM3NetworkPrinting
2018-11-27 09:32:56,254 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2018-11-27 09:32:56,258 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UpdateChecker
2018-11-27 09:32:56,296 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin USBPrinting
2018-11-27 09:32:56,316 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UserAgreement
2018-11-27 09:32:56,340 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade21to22
2018-11-27 09:32:56,361 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade22to24
2018-11-27 09:32:56,383 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade25to26
2018-11-27 09:32:56,404 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade26to27
2018-11-27 09:32:56,426 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade27to30
[... 562 lines cut ...]
2018-11-27 09:37:17,488 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6_/resources/qml/Topbar.qml:175: TypeError: Cannot read property 'stageId' of null
2018-11-27 09:37:17,490 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6_/resources/qml/Cura.qml:444: TypeError: Cannot read property 'mainComponent' of null
2018-11-27 09:37:17,491 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6_/resources/qml/Cura.qml:438: TypeError: Cannot read property 'mainComponent' of null
2018-11-27 09:37:17,492 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6_/resources/qml/Cura.qml:476: TypeError: Cannot read property 'sidebarComponent' of null
2018-11-27 09:37:17,493 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6_/resources/qml/Cura.qml:508: TypeError: Cannot read property 'sidebarComponent' of null
2018-11-27 09:37:17,499 - DEBUG - [MainThread] UM3NetworkPrinting.src.DiscoverUM3Action._createAdditionalComponentsView [171]: Creating additional ui components for UM3.
2018-11-27 09:37:17,530 - DEBUG - [MainThread] UM3NetworkPrinting.src.DiscoverUM3Action.startDiscovery [43]: Starting device discovery.
2018-11-27 09:37:17,539 - DEBUG - [MainThread] UM.Controller.setActiveStage [159]: Setting active stage to PrepareStage
2018-11-27 09:37:17,675 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [34]: Updating NozzleModel.
2018-11-27 09:37:17,735 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6_/resources/qml/SidebarHeader.qml:551:9: QML QQuickItem: Binding loop detected for property "width"
2018-11-27 09:37:17,778 - DEBUG - [MainThread] UM.Controller.setActiveView [92]: Setting active view to SolidView
2018-11-27 09:37:17,788 - DEBUG - [MainThread] ModelChecker.ModelChecker._createView [97]: Creating model checker view.
2018-11-27 09:37:17,796 - DEBUG - [MainThread] ModelChecker.ModelChecker._createView [106]: Model checker view created.
2018-11-27 09:37:17,798 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._createDialog [82]: Creating dialog [MoreInfoWindow.qml]
2018-11-27 09:37:17,843 - DEBUG - [MainThread] cura.CuraApplication.run [741]: Booting Cura took 8.530293703079224 seconds
2018-11-27 09:37:17,902 - INFO - [MainThread] cura.Machines.MaterialManager._updateMaps [279]: Updating material lookup data ...
2018-11-27 09:37:17,930 - INFO - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1226]: Updating quality/quality_changes due to material change
2018-11-27 09:37:17,932 - DEBUG - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1233]: Current quality type = [normal]
2018-11-27 09:37:17,934 - INFO - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1247]: Current available quality type [normal] is available, applying changes.
2018-11-27 09:37:17,935 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
[... cut ...] QualityChanges lookup table
2018-11-27 09:37:17,938 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [50]: Updating QualityProfilesDropDownMenuModel.
2018-11-27 09:37:17,965 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-11-27 09:37:17,967 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [34]: Updating NozzleModel.
[3.6.zip](https://github.com/Ultimaker/Cura/files/2620923/3.6.zip)

akf-was-here, 2018-11-28:
Ok, any idea what I do next?  I've uninstalled / reinstalled to a new directory but the behavior doesn't change.

Thierry-CR10S-user, 2018-12-05:
Hi, same thing for me. But i find why in my case: after upgrade V3.5.0 => V3.6.0 when trying to open my stl file by double click, the crash occur. When you open cura and next open your file it's OK
It's because windows try to open the file why Cura 3.5 
Only do "open why" on stl's file and choose the right version (careful you have 2 choices and the both why "Cura" name

akf-was-here, 2018-12-05:
Seems like not really the same thing then.  In my case I'm trying to directly open the Cura application (via it's own icon in it's own install directory) and it crashes where in your case that operation seems to work.  I'm not spawning Cura via opening an STL which in your case seems to be crashing.

Aiddy81, 2018-12-16:
I had the same issue where it wouldn't load after clicking the icon, I found it was due to an incompatibility with one of the settings from 3.5.1, on the crash window I selected backup and reset config. After doing this 3.6 loaded and I re-configured it.

akf-was-here, 2018-12-16:
My crash window is a Windows 10 crash window, there isn't any option to backup and reset config.

vegar, 2018-12-18:
I experience the same thing. Splashscreen shows, stops at 'Loading Interface...'. No cpu usage. 
Using SysInternals Process Monitor (see attached log), I can see that there is no more activity after loading some QtQuick controls. 

[ProcMonLogfile.txt](https://github.com/Ultimaker/Cura/files/2692550/ProcMonLogfile.txt)

steevegldev, 2019-01-13:
I got the same issue cause by my multi monitor setup. If I move Cura's window to a secondary monitor and click on some menu, Cura crash. When opening it again, it stuck at loading...

Workaround: 
1. Open the config file "%APPDATA%\Roaming\cura\3.6\cura.cfg"
1. In the section [general], remove every entry starting with "window_" (Those information are related to the window position).
1. Start Cura with success.

This will not happen again if the window is not move to the secondary monitor.

Be able to reproduce in: 
- Cura 4.0 beta
- Cura 3.6.0
- Cura 3.5.1

nallath, 2019-01-14:
It seems like this is an upstream bug from Qt (the graphical framework that we use).

JonS9999, 2019-01-28:
Hi.  I'm having similar issues: Cura 3.6.0 crashes at startup -- after the splash screen finishes, I see the Change Log window briefly, then a Windows pop-up appears and says that Cura has crashed.

I did some investigation a couple weeks ago, and it seems like it might be an issue with Python or with Qt.

Should I open a new trouble ticket, or should I chime in here since Qt was mentioned...?

-Jon

Ghostkeeper, 2019-01-29:
Seems like almost everyone in this thread is experiencing this problem slightly differently :laughing:

akf-was-here, 2019-01-30:
I hadn't opened my 3.6 install since my last post here (mid-Dec) but in trying steevegldev's suggested fix I found that it has "magically" fixed itself in the interim.   I've gotten several windows patches since then so maybe one of them altered the behavior.

MadderPad, 2019-02-27:
Just to add I got this with a box fresh Dell laptop today. Running as administrator fixed the issue. 

I suspect it's connected with the screen resolution/drivers/window layout - I have a 4k screen - running as my user account, the splash screen etc is way big (as if it's reading the screen resolution incorrectly), and it crashes at "loading interfaces" as described above. Running as Admin, the splash is the right size, and then loads as normal.

Paddy

JonS9999, 2019-02-27:
akf-was-here : I tried steevegldev's suggestion, but unfortunately it didn't help.

MadderPad : Interesting...  I have been using Cura 4.0 BETA successfully, so I'm hesitant to go back to 3.x (if I have to for some reason, I'll try your 'run as administrator' suggestion).

kurisun, 2019-03-13:
Ran into the issue on Windows 10 as well.  Trying to drag the Cura 3.6 window to my other monitor and then interacting with menu options causes a Not Responding error.  After closing Cura forcibly, and rebooting, the issue persists on subsequent launches.  I didn't know about the Cura window fix and instead disabled my second monitor before launching Cura, which fixed the issue.  I'm running the latest Nvidia drivers if that matters.

Ghostkeeper, 2019-03-13:
We're not really doing anything special other than creating a window with Qt and giving it the position that the last window was closed at. I have no idea how to approach this bug... :disappointed:

digigram, 2019-07-27:
@Ghostkeeper as a band-aid, maybe Uranium can be patched where it checks if the geometry is out of range, and constrain it to screen 0 only. I get this same bug any time more than 50% of the Cura window is on screen 1. Crashes consistently whether the window_left is positive or negative (i.e. screen 1 is to the left or right of screen 0). Both v3.6 and v4.1

Ghostkeeper, 2019-07-29:
We don't know which screen is screen 0 though.

I think that your problem manifests itself on other computers with different screens as well. It's not necessarily related to being on screen 0 or screen 1.

To limit Cura to screen 0 would be very annoying to users with multiple screens. For instance at work I use a laptop provided by Ultimaker with a tiny screen but I practically only use the plug-in monitor with it.

JonS9999, 2019-07-30:
Hi all.

I think I figured out why it's been crashing for me (I'm not saying this is everyone's reason; just mine).  I hope I can explain this well enough as I haven't had my morning coffee yet today...

I have two video cards in my PC -- one built-in video card (screen 1) and a PCI-Express video card (screen 0).  The PCI-Express is powerful enough to support a new enough version of Open-GL so that Cura layer-preview (sorry, not the right words -- it's the thing at the bottom of the Preview page that has a "play" button and will show you how each layer will be extruded).  My built-in video card is not powerful enough to run a newer version of Open-GL, thus, it won't do the layer-preview.

By default, Cura opens on screen 0 (PCI-Express card), and life is good.  But when I drag Cura to screen 0 (built-in video card), Cura crashes.

What clued me in to this was that I have an identical PC (same make & model, same amount of RAM, etc), *however* I don't have a PCI-Express video card in that PC -- I am only using the built-in video card.  I noticed that when I run Cura on that PC, I don't see layer-preview thingie.  Cura runs fine, but I just don't have the option of simulating the printing of layers.

So, that made me wonder if the problem is:

   - Moving the Cura window from a more powerful video card to a less powerful video card.

   - Moving the Cura window from one version of Open-GL to a different version of Open-GL.

Anyway, food for thought...

Cheers.

-Jon

akf-was-here, 2019-07-30:
@Jon: I have two identical pci express video cards (driving 5 monitors in my setup) in a similar config as your are describing while still having issues so I don't think (for me at least) it's a more to less powerful video system as you describe in your #1 above.  Mine are also both using the same drivers (I hope) so (again for me at least) that means it's not an Open-GL version difference either.

JonS9999, 2019-07-30:
> @jon: I have two identical pci express video cards (driving 5 monitors in my setup) in a similar config as your are describing while still having issues so I don't think (for me at least) it's a [...]

Well, so much for my theory -- LOL.  :-)  :-)

-Jon

Ghostkeeper, 2019-07-31:
Could you go into the preferences and enable "Force Layer View Compatibility Mode"? That should configure Cura to use OpenGL2.1-compliant shaders.

If you're unable to start Cura, you can set this setting by opening the file `C:\Users\<username>\AppData\Roaming\cura\4.2\cura.cfg` with a text editor. Under the category `[view]` you can add the following line:

    force_layer_view_compatibility_mode = True

closed 2024-12-08 by github-actions[bot]