Ultimaker/Cura #4897: Cura 3.6.0 , material override error in part of the gui,  Ultimaker3

UltiArjan, 2018-11-27:
[image]

**Application Version**
3.6.0 
**Platform**
Win 10

**Printer**
Ultimaker3 (also applies to S5)

**Steps to Reproduce**
Send a print to the printer with a different material selected in CURA vs the material selected on the machine.  In the screendump example it's generic CPE vs WHITE CPE, but it's the same for any different material.

When you select "override" directly from cura monitor (left part of screenshot) you get a override button, but pressing this actually does nothing.  

**Actual Results**
nothing, no error, no message, it just does not work

**Expected results**
The printer should start printing with the settings of cura and ignore the material selected on the printer.

**Additional Information**
The override DOES WORK when opening the "manage queue" and override the print job there. So the functionality is there, just the GUI / UX is broken. (right part of the screendump)

UltiArjan, 2018-11-27:
to clarify, the override buton does provide a dialogue box, but confirming override there does nothing

diegopradogesto, 2018-11-28:
ping @ianpaschal

ianpaschal, 2018-11-28:
Interesting. Pretty sure we would have caught that in QA/smoke test/etc. Friday with the new sprint I'll be re-writing this stuff for Cura 4.0 and will take a look into it. Thanks for the ping @diegopradogesto