Ultimaker/Cura #4898: Application form and mouse X,Y positioning is off

Jollym, 2018-11-27:
**Application Version**
3.6 (Just downloaded & installed it 2018/11/27)

**Platform**
Windows 10 Enterprise 64 (Just installed it 2 days ago)

**Printer**
Ender 3

**Steps to Reproduce**
Constent

**Actual Results**
ALL forms / mouse positions are negitivly hight offset

**Expected results**
All forms and windows and mouse positions should start at 0 , 0

**Additional Information**

I have tried stopping and restarting the app, I have tried rescaling the desktop, I have tried rebooting, nothing fixes the issue

[image]
[image]
[image]

fieldOfView, 2018-11-27:
Looks like https://github.com/ultimaker/cura/issue/2068

In short: update your intel gpu driver.

diegopradogesto, 2018-11-28:
@Jollym can you please confirm that updating the drivers fixes the issue? Thanks.

Jollym, 2018-11-28:
[image]

Confirmed Fixed.

NOTE: Originally #2068 was returning a 404 error but now the link is working.

fieldOfView, 2018-11-28:
Sorry, I made a typo in the url to #2068 when I initially posted my answer. I fixed it after the notification you received was sent.

fieldOfView, 2018-11-28:
Please close the issue.

closed 2018-11-28 by Jollym