Ultimaker/Cura #4900: Engine: Build error based on 3.6 branches

thopiekar, 2018-11-28:
Hey! :wink: 

There is a build error, when I build the CuraEngine based on 3.6 branches in the PPA. [1]
I personally can't follow on this error, that something "is marked as override, but not overwritten" (?!).
Updated Arcus and made sure protobuf is updated before the build, too, but just ran into this.
On the other hand the master PPA works fine, so it looks like there might be something that needs to be cherry-picked(?).

Cheers!

----------------

[1] https://launchpadlibrarian.net/399397969/buildlog_ubuntu-bionic-amd64.cura-engine_1%3A3.6.0~201811281133~rev2142~pkg278~ubuntu18.04.1_BUILDING.txt.gz

rburema, 2018-11-29:
Hi @thopiekar,
What do you mean when you say 'updated libArcus'? I ask because libArcus has a 3.6-tag as well, and if you've checked out libArcus master (or 3.5.1, or...) it might not work.
(Also, that error seems to be in the tests... while that is of course a problem, especially for something tagged release(!), the rest should still be able to compile, if the building of tests is disabled. Depending on what you whant to do, you might want to go that route.)
cheers, Remco.

LipuFei, 2018-11-29:
@thopiekar It's caused by some changes in libArcus. Could you check if the libArcus version dependency is correct?

thopiekar, 2018-11-29:
Yes, by updated I mean using the 3.6 branch. libCharon only misses a tag.
So if I'm not wrong I only use the master codebase for libCharon and this
should not affect the engine.

thopiekar, 2018-11-29:
Yes, can double check. Might be that I mixed up Charon with Arcus before...

thopiekar, 2018-11-30:
Checked and libArcus is built using 3.6 branch:
[image]

BagelOrb, 2018-12-07:
I also had a compile problem relating to libArcus. Somehow deleting the whole curaengine build dir fixed it.

thopiekar, 2018-12-09:
Hmm, yeah, but in my case, the builds are always made on a clean minimalistic installation.
I have the feeling that either Arcus or CuraEngine have been tagged wrong. But don't see how.. I mean checked the tags quickly and all seems to be good.

How are the plans for the following release? Would be great if I could get out of this situation, but upgrading the stable PPA to 3.7. I doubt you (Ultimaker) are willing to fix anything here anymore, right?

closed 2019-04-04 by nallath