Ultimaker/Cura #4902: Ultimaker 3 Active Leveling Error

RichardNoakes, 2018-11-28:
Ultimaker 3 firmware 4.3.3.20180529

Procedure: System > Build Plate > Active Leveling

Error: Difference between detected height of both PrintCores exceeds realistic values.

Notes: The bed has been manually leveled. The Active Leveling process appears to run through normally but reports the error on completion. Many repeated attempts produce the same result.

Any advice would be appreciated.

rburema, 2018-11-29:
pinging @Ultimaker/firmware

CoenSchalkwijk, 2018-11-29:
@RichardNoakes Please contact your re-seller for support with your issue.

closed 2018-11-29 by ChrisTerBeke

ckielstra, 2018-11-29:
As extra info:  check the position of your build plate leveling screws. These screws should be about halfway their minimum and maximum positions.
More info: https://ultimaker.com/en/resources/47734-difference-between-detected-height-of-both-printcores-exceeds-realistic-values

RichardNoakes, 2018-11-30:
@ckielstra - Thank you. I have checked the build plate screws and adjusted them as per the instructions.
If I disable the Active leveling function completely and do a manual leveling then I get good results.
The Active leveling procedure seems to vary slightly in that the build plate does not always travel to the same height, at some points it makes contact with the print nozzle, at other times not. This happens with both Nozzle 1 and Nozzle 2. Sometimes the build plate stops about 5 mm clear of the nozzle and reports the error after the procedure has ended.
The Active Leveling function worked before, I'm not 100% sure but I think it failed for the first time after I did a firmware upgrade three days ago.

ckielstra, 2018-12-04:
We get occasional reports from printers that fail in a way you describe but are impossible to reproduce when such a printer is shipped to us. The sensor used for the active leveling is an extremely sensitive capacitance sensor. What we suspect is that something in the printer's environment has changed, not something obvious, we checked those issues. And this external change introduces enough noise for the leveling sensor to fail.
Your best chance is to wait for the upcoming firmware release, number to be decided, but most likely v5.1. Due End of January 2019. This new version has an improved algorithm for filtering out noise in the Active Leveling. The same software is already used in the S5 and yields better results.

RichardNoakes, 2018-12-04:
@ckielstra - Thank you. Your comment makes complete sense. For now I can continue with manual leveling and leave the active leveling disabled. Your assistance is much appreciated.