Ultimaker/Cura #4909: Extruder Docking in Multi-Extruder Configuration

UticaTechClub, 2018-11-29:
Dear Cura Development Team,
Thank you so much for a good product and for keeping it accessible in open source realm.

**Application Version**
Cura 3.6.0

**Platform**
Windows 10, 64 bit

**Printer**
Custom FDM multi-extruder.
Four extruders docked on the back wall. 

**Steps to Reproduce**
Print any multi-material model.

**Actual Results**
Last used "Extruder End G-code" is not executed.
Last used extruder remains attached to the tool holder.

**Expected results**
Last used "Extruder End G-code" should be executed.
Last used extruder would be properly docked to its' slot on the back wall.

**Additional Information**
On T<N> G-code command (where <N> is extruder index), N's "Extruder Start G-code" is executed and that code knows how to navigate to N's tool dock and attach it to the tool holder. At the end of the tool usage, N's "Extruder End G-code" is executed and it knows how to navigate to N's dock and disconnect it from the tool holder.

Cura should execute last used "Extruder End G-code", because only that code knows where the last extruder needs to be docked at.

I am a aware that Cura always executes "End G-code", but that code can not be conditioned on the docking location of the last used extruder. At least, I am not aware that such conditional statements exist.

I am also aware that in multi-extruder case, Cura also assumes T0 to always be the first and will not execute the T0's "Extruder Start G-code", but in that case I can at least inject T0's "Extruder Start G-code" into "Start G-code" knowing it is specific for T0.

I don't know why wouldn't it make sense to always execute "Extruder Start G-code" and "Extruder End G-code" even on single extruder printers. Developers could always leave those sections blank if no extruder specific action is required.

Thank you and kind regards.

UticaTechClub, 2018-11-30:
**To fix my problem, one of two things needs to happen:**
   **a)** Figure out if Cura could invoke "Extruder End G-Code" on the last used extruder. I am sure there is a good reason(s) as to why Cura doesn't do it.
   **b)** Figure out how to condition gcode invoked in "End G-Code" section. After much reading, I believe this is currently not possible in Cura.

UticaTechClub, 2018-11-30:
**Perhaps there is another way to make "End G-Code" conditional as found in**
[https://community.ultimaker.com/topic/21314-possible-to-make-dynamic-gcode-in-cura/](url)

The idea is to define value property in (.def.json) and embedd Python expression to be evaluated like in the following example:
`"machine_end_gcode": { "value": "'G29 P' + str(int(machine_width/25))" },`

So in my case, I could pass proper X,Y offset to G1 based on {current_extruder} property.

I think it will work...

UticaTechClub, 2018-12-04:
I just found this topic describing exactly same behavior:
### [https://github.com/Ultimaker/Cura/issues/3269](url)

I humbly suggest that without properly handling of the first `machine_extruder_start_code` and the last `machine_extruder_end_code` it would be impossible to use Cura for any advanced multi tool configurations.

Please Cura Team, please take a look at this one.
@Ghostkeeper 
@nallath 

It sound like it should be low risk and transparent to almost all configurations (including Ultimaker) that do not use `machine_extruder_start_code` and `machine_extruder_end_code`.

rburema, 2019-06-07:
Hi @UticaTechClub, Out apologies it took so long to get back, this issue-report really slipped between the cracks :-(

As you can see from the related issue linked above, we've chosen to defer this, as the printers that need this are relatively rare, and right now, the extruder-end-code is only meant for switching between tools during a print. Changing this guarantee would mess up other processes. There are ways around this, but given the effort required this doesn't have priority right now. (Though we do welcome pull-requests.)