Ultimaker/Cura #4910: COASTING BEFORE LAYER CHANGE

fishonthefly, 2018-11-29:
**Application Version**

VERSION 3.6.0
**Platform**

WINDOWS 10
**Printer**

Creality3D CR-10
**Steps to Reproduce**

I'm not sure
**Actual Results**

I'm not sure which version of Cura this came up in. The settings I am currently using worked great at one point. Now I am getting thin/weak spots in my outer wall. I noticed today that the extruder was stopping feeding about 25mm before the layer change. I have coasting and combing turned off.
**Expected results**

The extruder should feed filament until the point that the layer change occurs so that it doesn't run out of material
[image]

**information**

[CCR10_TINY TRAINER EDGE GUARD.zip](https://github.com/Ultimaker/Cura/files/2630958/CCR10_TINY.TRAINER.EDGE.GUARD.zip)
[CCR10_TINY TRAINER EDGE GUARD.zip](https://github.com/Ultimaker/Cura/files/2630959/CCR10_TINY.TRAINER.EDGE.GUARD.zip)

smartavionics, 2018-11-29:
Hi @fishonthefly , I have sliced your project and cannot see any coasting before the layer change. Could you please zip up the gcode that printed the object in your photo and attach that for me to look at. Thanks.

smartavionics, 2018-11-29:
Sorry, just noticed that you have already done that - ignore my previous reply!

fishonthefly, 2018-11-29:
I ran the print again and watched closely. It looks like the weak line is occurring on the new layer. As if it is taking a while for the nozzle to prime again after the layer change. I have an extra prime amount and speeded up the prime speed, so I am not sure why it is doing this. The extruder does still stop feeding before it gets to the end of the line though, as if coasting is enabled.

smartavionics, 2018-11-29:
OK, I think I know what is happening. The inner wall is being flow reduced to almost nothing by the overlap compensation and so the extruder will be dribbling while the nozzle moves from the other end of the part to the end where the layer starts. There is a setting called Minimum Wall Flow, set that to some non-zero value like 20 and it should replace that very thin inner wall with a travel move which will use retraction and so the nozzle won't be empty when it starts the next layer. Hope this helps.

fishonthefly, 2018-11-29:
I am giving that a try as I write this. I have been trying to make sure that my walls have an even line count when I design things. But that proves to be tricky. For instance, in this design, I wanted a wall three lines thick, so I started with a 1.2mm thickness. The slicer made the inner line zig zags, so I kept adding thickness by .1 until it made the inner line a straight line. Do you have any suggestions on this?

fishonthefly, 2018-11-30:
The solution Mark suggested in his last post fixed the problem. 
Thank you Mark.

closed 2018-11-30 by fishonthefly