Ultimaker/Cura #4912: Crash when opening the Marketplace

MightyCreak, 2018-11-30:
**Application Version**
3.6.0 (reproducible with 3.5.1 and 3.5.0)

**Platform**
Linux (Fedora 29)

**Printer**
Anycubic Kossel Linear Plus

**Steps to Reproduce**
In the menu: Marketplace > Marketplace
(happens all the time; tried 10 times, 10 times it failed)

**Actual Results**
Crash

**Expected results**
Show the Marketplace

**Additional Information**
Here are the lines from the output once I clicked on the menu item (don't know if it helps):
```
Auto configuration failed
139684462712640:error:0E079065:configuration file routines:DEF_LOAD_BIO:missing equal sign:conf_def.c:362:line 40
QMutex: destroying locked mutex
```

MightyCreak, 2018-11-30:
Yeah I wasn't sure, but I thought it was working before (on Fedora 28) as well.

If you're right about SSL, it remembers me the last time I had the same kind of problem with Mercurial: the crypto algorithm used was too old and got removed from the Fedora lib. Maybe it's the same here?

nallath, 2018-11-30:
Well, we're using https certificates. I doubt that those are suddenly dropped from Fedora.

morozsm, 2018-12-20:
I have exactly the same issue.  Any workaround by this time?

nallath, 2018-12-21:
Nope, not yet. We haven't been able to figure out a way to resolve this without also breaking other things.

morozsm, 2018-12-21:
There is workaround on Ubuntu. Setup OpenSSL1.0.0 compatibility package and dev package. However, it doesn't work with Fedora29.
Curs is distributed via flatpack. Isn't it possible to place appropriate version of LibSSL to flatpack package?

nallath, 2019-01-03:
We fixed the issue for the second beta of 4.0.0

closed 2019-01-03 by nallath

Yannik, 2019-02-16:
Here is how to fix this for 3.6.0:
`sudo dnf install compat-openssl10`

Start cura like this:
`LD_PRELOAD="/usr/lib64/libssl.so.10 /usr/lib64/libcrypto.so.10" ./Cura-3.6.0.AppImage`