Ultimaker/Cura #4913: Local axis helpers to translate/rotate/scale objects

MightyCreak, 2018-11-30:
**Application Version**
3.6.0

**Platform**
Linux (Fedora 29)

**Printer**
Anycubic Kossel Linear Plus

**Steps to Reproduce**
Add a support blocker
Rotate it on Z axis
Try to translate it on the local X axis

**Actual Results**
You can't, you only have access to the global axis

**Expected results**
Being able to switch between local and global helpers

**Additional Information**

alekseisasin, 2018-12-05:
I added a support block and I was able to rotate it in X-, Y- and Z-coordinates. I think this is what you mean global coordinates. When I rotate I get expected position.
How will it help you using different (local and global) coordinates?

MightyCreak, 2018-12-05:
I was working on a mini and the placement of these support blocks needs to be quite precise. For instance I added a block, reduced it, scaled it along X and aligned it under the sword of my mini. Having access to local translation helpers would have helped me placed it more precisely since the sword was not aligned with the world axes.

But that's not all, the block was not long enough to cover the whole sword, but since I can't scale on the local axes, I had to rotate the block back into a world alignement, do the scale on the global X axis, and re-rotate it back under the sword.

If it was possible to switch between local and global, I would just have to switch to local, scale on X, and voila.

The local/global thing is very common in 3D tools (3DS max, Maya, Blender, Unity, ...)

alekseisasin, 2018-12-06:
I discussed your request and unfortunately, we cannot implement it. Cura team was working on something similar and used to stop because it has a lot of edge cases which requires significant effort.

For me, the workaround could be to make a support block in another CAD program and load to Cura.  Yes, it will cost time but if I would print the same model I could be able to reuse it. It is not the best proposal.

MightyCreak, 2018-12-06:
It's a bit disappointing, but I understand. I'll try to use another CAD program. Thanks!