Ultimaker/Cura #4914: 3.6.0 seems to flash UMO+ with the wrong firmware

LarsSimonsen, 2018-11-30:
Cura 3.6 seems to feed the wrong firmware to the UMO+, temporarily bricking it until the correct firmware is uploaded using an older version of Cura.

**Application Version**
3.6.0

**Platform**
Windows 10 64 bit

**Printer**
UMO+

**Steps to Reproduce**
Connect UMO+ to Windows10 via USB. Go to printers, select the UMO+ and "Automatically upgrade firmware".

**Actual Results**
Printer becomes braindead. Blank screen, steppers held in place. Initializing homing axis from Cura causes noise until printer is power cycled.

**Expected results**
Firmware flash should be successful or error message should appear in Cura.

**Additional Information**
Seems to be a known issue of Cura offering the wrong firmware to the UMO+.
https://community.ultimaker.com/topic/25562-umo-firmware-problem-with-cura-360

nallath, 2018-12-03:
Devs, see CURA-5994

LipuFei, 2018-12-18:
This is fixed in 4.0 via https://github.com/Ultimaker/Cura/pull/5015

closed 2018-12-18 by LipuFei