Ultimaker/Cura #4915: Cura 3.5.0 increasing print speeds over time

charles1024, 2018-12-01:
**Application Version**
3.50

**Platform**
Windows 10

**Printer**
Ender 3

**Steps to Reproduce**
Print several models over the course of 6-18 hours. Print speed will gradually increase, even thought its set to 60mm/s.
https://youtu.be/q0pR8eefQ58?t=110

**Actual Results**
Ender 3 prints at ludicrous speed

**Expected results**
Ender 3 prints at a normal 60mm/s

**Additional Information**

I have noticed this behavior regardless of the project or model I load.

I am a software engineer by trade. This is legit one of the most bizarre bugs I have ever come across. I have tried various things to ensure I can re-produce this.  I have included the linked video to show whats going on. The only thing that 'postpones' the ridiculous print speed increase is regularly restarting Cura and doing a full power cycle on the Ender 3.
https://youtu.be/q0pR8eefQ58?t=110

Cura Config:
[info]

[backend]

[general]
visible_settings = support_tree_branch_distance;travel_avoid_distance;prime_tower_position_y;speed_print;material_bed_temperature_layer_0;wall_line_count;retraction_prime_speed;support_tree_angle;retraction_combing;retraction_hop;platform_adhesion;top_thickness;wall_thickness;layer_height;support_tree_branch_diameter_angle;adhesion_type;travel_avoid_other_parts;support_infill_rate;wall_min_flow_retract;meshfix;machine_settings;coasting_enable;support_enable;top_bottom_thickness;prime_tower_position_x;remove_empty_first_layers;material_bed_temperature;retraction_retract_speed;cool_fan_speed;support_extruder_nr;retraction_hop_only_when_collides;layer_height_0;acceleration_travel;support_tree_enable;travel;dual;support_pattern;speed_travel;retract_at_layer_change;speed;cooling;speed_wall;material;top_layers;acceleration_print;support_use_towers;prime_tower_enable;infill_sparse_density;material_flow;cutting_mesh;raft_speed;prime_blob_enable;xy_offset;infill_pattern;support;support_type;ironing_enabled;shell;speed_infill;cool_fan_enabled;support_skip_some_zags;retraction_enable;experimental;adhesion_extruder_nr;resolution;retraction_amount;infill;bottom_layers;material_print_temperature_layer_0;blackmagic;material_print_temperature;support_angle;retraction_hop_enabled;support_tree_branch_diameter;retraction_speed;adaptive_layer_height_enabled;bottom_thickness;conical_overhang_enabled
window_height = 819
latest_version_changelog_shown = 3.5.0
accepted_user_agreement = True
window_state = 2
window_left = 118
window_top = 23
window_width = 1582
version = 6

[um3networkprinting]

[local_file]
dialog_load_path = D:/3D Printing/mom/Deer_Wall_Sculpture_2D_II/files
dialog_profile_path = D:/3D Printing/siepie_debug
last_used_type = application/vnd.ms-package.3dmanufacturing-3dmodel+xml
dialog_save_path = D:/3D Printing/parts/Ender_3_Cable_Chain_

[metadata]
setting_version = 5

[physics]

[view]

[cura]
dialog_on_project_save = False
recent_files = D:/3D Printing/mom/Deer_Wall_Sculpture_2D_II/files/deer2.stl;D:/3D Printing/parts/Ender_3_Cable_Chain_/files/cableChain_XendCover.stl;D:/3D Printing/parts/Ender_3_Cable_Chain_/files/cableChain_Xend.stl;D:/3D Printing/parts/Ender_3_Cable_Chain_/files/cableChain_plateMount_v2.stl;D:/3D Printing/parts/Ender_3_Cable_Chain_/files/cableChain_bedCorner.stl;D:/3D Printing/parts/Ender_3_Cable_Chain_/files/cableChain_link_v3.stl;D:/3D Printing/parts/Ender_3_Cable_Chain_/files/cableChain_bedCorner_v2.stl;D:/3D Printing/parts/Ender_3_Cable_Chain_/files/cableChain_link_v2.stl;D:/3D Printing/parts/Ender_3_Cable_Chain_/files/cableChain_bedCornerCover_v2.stl;D:/3D Printing/parts/Ender_3_Cable_Chain_/files/cableChain_40MountTop_v2.stl;
expanded_brands = ;Generic
categories_expanded = ;acceleration_layer_0;acceleration_print;acceleration_support;acceleration_support_interface;acceleration_wall;bottom_thickness;brim_width;cool_fan_full_at_height;cool_fan_speed;cooling;dual;expand_skins_expand_distance;experimental;infill;infill_overlap;infill_sparse_density;jerk_layer_0;jerk_print;jerk_support;jerk_support_interface;jerk_wall;line_width;magic_fuzzy_skin_point_density;material;max_skin_angle_for_expansion;meshfix;platform_adhesion;raft_acceleration;raft_fan_speed;raft_jerk;raft_speed;resolution;retraction_speed;shell;skin_overlap;skin_preshrink;speed;speed_layer_0;speed_print;speed_support;speed_support_interface;speed_wall;support;support_bottom_density;support_extruder_nr;support_infill_rate;support_interface_density;support_interface_enable;support_interface_extruder_nr;support_interface_height;support_interface_line_width;support_interface_pattern;support_roof_density;support_skip_zag_per_mm;support_tree_wall_thickness;support_z_distance;switch_extruder_retraction_speeds;top_bottom_thickness;top_thickness;travel;travel_compensate_overlapping_walls_enabled;wall_extruder_nr;wall_line_width;wall_thickness;wireframe_flow;wireframe_printspeed
custom_visible_settings = support_tree_branch_distance;travel_avoid_distance;prime_tower_position_y;speed_print;material_bed_temperature_layer_0;wall_line_count;retraction_prime_speed;support_tree_angle;retraction_combing;retraction_hop;platform_adhesion;top_thickness;wall_thickness;layer_height;support_tree_branch_diameter_angle;adhesion_type;travel_avoid_other_parts;support_infill_rate;wall_min_flow_retract;meshfix;machine_settings;coasting_enable;support_enable;top_bottom_thickness;prime_tower_position_x;remove_empty_first_layers;material_bed_temperature;retraction_retract_speed;cool_fan_speed;support_extruder_nr;retraction_hop_only_when_collides;layer_height_0;acceleration_travel;support_tree_enable;travel;dual;support_pattern;speed_travel;retract_at_layer_change;speed;cooling;speed_wall;material;top_layers;acceleration_print;support_use_towers;prime_tower_enable;infill_sparse_density;material_flow;cutting_mesh;raft_speed;prime_blob_enable;xy_offset;infill_pattern;support;support_type;ironing_enabled;shell;speed_infill;cool_fan_enabled;support_skip_some_zags;retraction_enable;experimental;adhesion_extruder_nr;resolution;retraction_amount;infill;bottom_layers;material_print_temperature_layer_0;blackmagic;material_print_temperature;support_angle;retraction_hop_enabled;support_tree_branch_diameter;retraction_speed;adaptive_layer_height_enabled;bottom_thickness;conical_overhang_enabled
active_setting_visibility_preset = custom
active_mode = 1
active_machine = Creality Ender-3

[layerview]
extruder_opacities = 1

[gcodereader]

[mesh]

fieldOfView, 2018-12-01:
Is it due to Cura exporting gcode with increasing speeds (in which case it should be visible in the saved gcode) or is your printer firmware interpreting the gcode with increasing speeds?

Tonio1987fr, 2019-02-19:
I have the same problem with my anet a8 marlin 1.1.9.
She goes fast and blocks the engines. I do not have this problem with other software.
Do you have a solution ? thx

[CFFFP_xyzCalibration_cube.txt](https://github.com/Ultimaker/Cura/files/2880460/CFFFP_xyzCalibration_cube.txt)

Ghostkeeper, 2019-02-22:
Maybe we should add that to the start g-code then?

    M502 ;reset firmware settings
    M500 ;save firmware settings

closed 2020-12-11 by no-response[bot]