Ultimaker/Cura #4916: Shortcut of the road - defects in vase (spiralize) mode

DeltaPlus99, 2018-12-01:
**Application Version**
3.6.0 and earlier

**Platform**
Windows 10, 64 bit

**Printer**
FFF - generic Marlin ; (Creality CR-10 Mini)
[vase.zip](https://github.com/Ultimaker/Cura/files/2635747/vase.zip)

**Steps to Reproduce**

- Custom FFF printer ; 300x220x300, heated bed, Marlin
machine_nozzle_size = 0.4
material_diameter = 1.75

- Cura settings:
adhesion_type = skirt
layer_height = 0.2
layer_height_0 = 0.2
magic_spiralize = True
top_bottom_thickness = 0.8
wall_line_width_0 = 0.8
wall_thickness = 1.2

https://www.thingiverse.com/thing:2581807 - vase.stl
(and other files in 40_vases_stl.zip)

**Additional Information**

[image]

[image]

The occurrence of defects is in some way random. For example, in the vase.stl model, defects occur at 50%, 100% or 150% scaling, and they do not occur at 75% or 155% scaling. By changing various parameters in Cura, defects can disappear or reappear

It can be observed also on other models in vase (spiralize) mode. For example:
https://www.thingiverse.com/thing:550604 - SpiralVaseHD.stl  - in 100%

Those defects cause the printed model to not be watertight anymore.

[image]

smartavionics, 2018-12-01:
Hello @DeltaPlus99 , sorry about that. I have fixed this many months ago but the fix has not yet been merged into Ultimaker's version.

I can look all around the layer view and can't see any bad lines like in your images above.

[image]

If you wish to try my Cura, there are pre-built versions at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

DeltaPlus99, 2018-12-02:
Hello, @smartavionics. I installed your version Cura-mb-master-win64-20181130.exe.
It seems, that in your version there are no problems with models anymore in spiralize mode. Thanks.

smartavionics, 2018-12-02:
Great! Hopefully those changes will make it into the standard Cura sometime. If you are happy, please close this issue. Cheers.

DeltaPlus99, 2018-12-05:
It's a good news. Vases printed in spiralize (vase) mode will be watertight. :)
Thanks. I close this issue.

smartavionics, 2018-12-13:
Hello @DeltaPlus99 , if you have a problem with my Cura releases please raise an issue on the smartavionics repo and not the ultimaker repo.

BTW, setting the slicing tolerance to exclusive solves this issue. Also, I found it looked better if the spiral smoothing was turned off.

DeltaPlus99, 2018-12-13:
Actually, the problem is with Cura 3.6.0 - see my first post on this topic.
It seemed that the Cura-mb-master-win64-20181130.exe version fixes the problems in question, but this hope turned out to be premature.
I will be happy when the problem will not occur in the next major versions of Cura.

smartavionics, 2018-12-13:
Try setting the slicing tolerance to exclusive, that worked for me.

DeltaPlus99, 2018-12-13:
The issue still exists in both Cura versions.

Setting:
slicing_tolerance = exclusive

https://www.thingiverse.com/thing:2704779
Model Christmas_Tree_Hard_Edge_Tweak.STL
Scale - 120%
[Christmas_Tree_Hard_Edge_Tweak.zip](https://github.com/Ultimaker/Cura/files/2675836/Christmas_Tree_Hard_Edge_Tweak.zip)

[image]

smartavionics, 2018-12-13:
It's working OK for me using my mb-master Cura. If it doesn't work for you using that Cura release, please save the project file and create a new issue at https://github.com/smartavionics/CuraEngine/issues and attach the project file. Thanks.

DeltaPlus99, 2018-12-13:
The occurrence of defects is in some way random.  By changing scale of the model, defects can disappear or reappear. By changing various parameters in Cura, defects can disappear or reappear.

I checked that defects exist in both Cura 3.6.0 and smartavionics mb-master versions (Cura-mb-master-win64-20181130 i Cura-mb-master-win64-20181210). Also when we set slicing_tolerance = exclusive.

In my previous post there is an example when the defect occurs at the 120% scale, but it does not occur at the 100% scale. This project (zip file) comes from the version Cura-mb-master-win64-20181130 for settings slicing_tolerance = exclusive.

closed 2019-05-10 by rburema