Ultimaker/Cura #4917: Suggestion: Speed Control

rodny098, 2018-12-01:
-Add to adjust a parameter in percent that control all other speed parameters.

Example: 
Global speed: 100%

means:
print speed = print speed * global_speed
wall speed = wall speed  * global_speed
...
...

alekseisasin, 2018-12-04:
Wall speed is already affected by print speed (speed_wall = "speed_print / 2"). Then you need to apply _global speed_ for "_print speed_".

JohnEdwa, 2018-12-04:
Gets my vote, but it should just be Feed Rate, the G-code command [M220](http://marlinfw.org/docs/gcode/M220.html) It would then be also modifiable from the printer LCD during the print as well.
Adding it as the very first option under the "Speed" category would be the best in my opinion.

@kaleidoscopeit While true, if you want to tweak any of them and still have them linked to the global "Print Speed" setting, you have to do those edits to the "quality_changes" .cfg files manually. For example, Ender 3 sets Infill Speed as "=math.ceil(speed_print * 100 / 60)", which means "Print Speed" of 60mm/s sets Infill to 100mm/s, which is too much but all the other speeds are good. If I write "80" there, I can't change it with "Print Speed" anymore.

closed 2026-09-06 by GregValiant