Ultimaker/Cura #4918: Windows 10 Pro/ Cura 3.51/ 3.6 - text boxes do not show text right menu. display text above input boxes on the right menu.

M357Y, 2018-12-01:
**Application Version**
3.51 and 3.6

**Platform**
Windows 10 Pro. Java version 8 update 191 (build 1.8.0_191-b12) Geforce GTX 1070 with latest drivers. Direct X runtime 12.0. Geforce Driver version 417.01. Driver type    DCH
Tested resolutions. 1920x1200, 1920x1080,1680x1050,1600x1200,1600x1024,1600x1000,1600x900

**Printer**
Creality Ender 3 (but it doesn't matter for this issue.

**Steps to Reproduce**
Open Cura. Text input is displayed above the input boxes on the left menu (scale/ rotate/etc).Text input does not display in the right hand window. Drop down text still works in the right menu.

**Actual Results**
See above.

**Expected results**
Text should align with input boxes on the left and actually be visible on the input boxes on the right.

**Additional Information**
[image]

nallath, 2018-12-03:
Do you have any accessibility features enabled on your operating system?

M357Y, 2018-12-07:
I thing accessibility and ease of use are the same. Correct me if not. I just have default settings
[image]

M357Y, 2018-12-07:
I've read quiet a few threads about screen size and text with windows 10 pro but I haven't found a combination that fixes the "no text" issues

fieldOfView, 2018-12-07:
Have you tried this workaround?
https://github.com/Ultimaker/Cura/issues/4784#issuecomment-443146888

M357Y, 2018-12-07:
Thank you fieldOfView
The workaround was successful.
I really appreciate your help!

closed 2019-01-08 by nallath