Ultimaker/Cura #4919: Unable to slice due to print head clearance

six-edge, 2018-12-01:
**Application Version**
3.6.0

**Platform**
Windows 10 
[UMO_Marble_Run.zip](https://github.com/Ultimaker/Cura/files/2636344/UMO_Marble_Run.zip)

**Printer**
Ultimaker+

**Steps to Reproduce**
Put objects on bed where the print head area overlaps.
Gantry Height is not exceeded.

**Actual Results**
Unable to slice

**Expected results**
Cura should slice it but change the print order so that no collisions occur. In this specific case, start to print the objects on the far right first (see screenshot) to avoid print head with object collision. I'm pretty sure this was not the case with 3.5.x but I not certain. However I think this logic makes sense as long as the Gantry Height is not exceeded.

A workaround as I have done in the past would be to reduce the print head dimensions, however in that case it is very difficult to control the order of print so that no collisions occur. To solve this problem, a manual edit of the sequence through the GUI would be a nice solution... or an intelligent slicer that can work out the best route without collisions.

**Additional Information**
[image]

nallath, 2018-12-03:
This has been fixed for the next release. Cura 3.6 has a bug with one at a time and the head areas.

closed 2018-12-03 by nallath

six-edge, 2018-12-03:
Awesome! Thank you

six-edge, 2018-12-03:
Is there a pull request or has it been merged?

nallath, 2018-12-04:
It has already been merged.