Ultimaker/Cura #4921: Cura 3.6 flashes on screen then disappears

freebird1963, 2018-12-02:
**Application Version**

3.6.0
**Platform**

Win 7 x64
**Printer**

Generic
**Steps to Reproduce**

Go to programs
Click on cura
watch it open up
then watch it disappear
tried running as administrator. same results
**Actual Results**

nothing. no cura on the screen. nothing. blank screen.
**Expected results**

cura should be running on the screen
**Additional Information**

processes show curaengine.exe running at 4460K working memory
tried running as administrator, failed to load also. 
Ran when first installed but not since.

nallath, 2018-12-03:
Could you share your log files?

freebird1963, 2018-12-04:
Sorry I got annoyed with it and removed it from my system. I am going to download and install again to see if maybe something went wrong.If it continues on this new install I will send them.
Thanks

thucar, 2018-12-09:
I'm having the same issue with both 3.5.1 and 3.6.0

[cura.log](https://github.com/Ultimaker/Cura/files/2660326/cura.log)

closed 2020-06-20 by no-response[bot]

Ghostkeeper, 2020-06-30:
I think this issue has been fixed in 4.7 due to some defensive coding we've implemented to not use the `extruders` dictionary any more but access the list of extruders by index.