Ultimaker/Cura #4922: One-at-a-time reducing printable dimensions

stevenfayers, 2018-12-02:
**Application Version**
3.6.0

**Platform**
Windows 10

**Printer**
Creality CR-10

**Steps to Reproduce**
Set up machine printhead settings with, for example 30mm for Xmin Xmax, Ymin, Ymax
Set one-at-a-time for print sequence
Add 2 models to the build plate
Drag one model near to the edge of the buildplate, within the boundaries but close.

**Actual Results**
model near buildplate edge becomes hashed out as 'unprintable'

**Expected results**
model near buildplate ewdge should be printable. Printhead dimensions should only be considered between models, not at the buildplate boundary

**Additional Information**

nallath, 2018-12-03:
There was a bug with the disallowed areas for 3.6 that made them too big when printing one at a time. This should be solved for the next release (4.0)

closed 2018-12-03 by nallath

ezorita, 2019-08-07:
This bug is still present in Cura 4.2.1.

[image]

Please reopen the issue.

Ghostkeeper, 2019-08-08:
The disallowed areas should be bigger when printing one at a time, that is intended behaviour. It has two reasons:
- There is a collision area around each object related to the size of the print head, that prevents you from placing objects so close together that the print head would collide with previously printed objects while printing the later objects.
- The brim, collision avoidance ratio, etc. are counted with the object rather than around the build volume.

Your screenshot looks about normal to me.

stevenfayers, 2019-08-08:
Hi Ghostkeeper

The collision avoidance shouldn't prevent the object being placed close to
the edge of the build plate. Currently if any of the collision area is
outside of the build area this prevents the object from printing. So
collision outisde build area should not be considered a collision.

Best regards

ezorita, 2019-08-08:
Indeed, by printer construction the printhead must be able to move freely outside the build area as long as the object is entirely contained in it.

The collision area should only take into account the overlap with other objects.

Ghostkeeper, 2019-08-12:
Indeed. As far as I can see, it's allowing that fine.

[image]

There are two collision areas here though: One for the print head and one for the print itself. The collision area for the print itself (including the brim) is not allowed to go outside of the build volume.