Ultimaker/Cura #4923: Cura 3.6. Bug: Corrupted "machine_extruder_start_code":"default_value" when in tool path

UticaTechClub, 2018-12-03:
**Application Version**
3.6.0

**Platform**
Windows 10 64bit.

**Printer**
Custom FDM, Marlin 1.1.9 based.

**Steps to Reproduce**
Slice any multi-extruder job.

**Actual Results**
Printer runs well until tool change command and than stops printing.

**Expected results**
Expect to execute tool change command and continue printing.

**Additional Information**
My two extruder printer has custom start and end g-code for each extruder. Configuration is coming from .def.json configuration files. After job is sliced and g-code file is saved, T0 and T1 commands are surrounded by the corrupt code. Printer stops printing on the first encounter with corrupted line.

Please note, that Cura Machine Settings UI, shows custom commands coming from printer configuration files without any problems
[image]
[image]
[image]

.

All files are attached.
Please help.

UticaTechClub, 2018-12-03:
[image]

UticaTechClub, 2018-12-03:
Please also note that print job start and end g-code is inserted without any problems.
Only extruder g-code exhibits this behavior.

nallath, 2018-12-03:
I don't know where those characters suddenly come from, but it doesn't happen on my machine with any of the machines that we ship with Cura.

UticaTechClub, 2018-12-03:
@nallath 

Are you on the same platform/version?
Have you tried loading my configuration?
Could you point to your printer configuration with multi-extruder custom code so I can try it?

Thank you

nallath, 2018-12-04:
I'm not on the same OS (Fedora 28). But I'm saying that it's probably nothing that Cura is doing, otherwise, you'd see this behavior for all printers. 
I've also tried adding the files that you provided, and these extra characters don't show up.

UticaTechClub, 2018-12-04:
> I'm not on the same OS (Fedora 28). But I'm saying that it's probably nothing that Cura is doing, otherwise, you'd see this behavior for all printers.
> I've also tried adding the files that you provided, and these extra characters don't show up.

Than you so much for following up on this nallath.
This issue can now be closed.

**The root cause:**
Some pattern in g-code comments section caused parsing problem.
I was able to reproduce this issue on two windows machines, both running Cura 3.6.0, but after removing comments, the g-code was properly inserted by the Cura engine.

Thank you so much for all the good work you guys are doing!!!

closed 2018-12-04 by UticaTechClub

UticaTechClub, 2018-12-04:
I will send details in few minutes...