Ultimaker/Cura #4927: CURA 3.6.0 Monitor mode - No printing temperature for extruder 1

Zigoto12345, 2018-12-03:
Hi,
When I want to start printing with USB mode, nothing happend on the printer because cura doesn't send any temperature order for the extruder (0°C.
A temperature (205°C) is set for this material in the "prepare" window.

Everything is ok with CURA 3.5.0 but this problem occurs with CURA 3.6.0 everytime.

I am using DAGOMA DISCOVERY printer. 

I tried to uninstall, clean the config files and reinstall CURA 3.6.0. The issue persists.
When I reinstall CURA3.5.0 = Ok

Fabrice

javicarrera, 2018-12-04:
Hello,
Exact same problem here, with Delta printer, just went back to the previous 3.5.1 and everything perfect.
Besides extruder's temperature not working, with the new version does not even detect that it has hot bed. And to conclude, no communication via gCode is possible.

Javier

CasLoL, 2018-12-18:
Hi,
like mentioned by Javier, my Vertex K8400 doesn´t work with Cura 3.6 either. Downgraded to 3.5.1 which works fine. Sadly I can´t try the new gyroid infill. I hope this Issue gets fixed.
CasLoL

grundsch, 2019-02-04:
Same here, Vertex K8400 doesn't work with Cura 3.6, but does with 3.5.1.
It's not only the temperature readout which doesn't work, but all G code too. I tried to have a look at the serial communication with a serial sniffer, and it looks like that cura 3.6 is using a different communication driver or timings. The answers to the M105 (temp read) are coming in chunks , interleaved by other M105 (as if it was issuing these commands without waiting for an actual answer). After issuing ca 6x M105, it issues a M115 (get firmware version) and get stuck, never receiving an answer from marlin.
With 3.5.1, the M105 messages are coming every ~3 sec instead, and the M105(x0a)M115 (sent at the same time) answers are coming fine.
It's not really clear cut, the traces are messy, but it's seems to be the main difference between the two traces...
Hope it helps...

Ghostkeeper, 2019-02-05:
I can't reproduce this using an Ultimaker 2, not in 3.6 nor 4.0.
[image]

I think that if there is a bug here, it'll be dependent on the printer and I can't test with your printer.

Slipery, 2019-03-31:
Same problem here using vanilla Marlin 1.1.9.  Works ok from sdcard but not USB

Ghostkeeper, 2019-04-01:
You get temperature feedback when printing with an SD card? That's telepathy :open_mouth: 

To debug this we'll need to know how your firmware reacts differently to the `M105` command, to see if Cura is interpreting it wrongly.

fieldOfView, 2019-04-01:
You can print from SD Card and have USB connected. I would not recommend it (because restarting Cura will abort the print), but it would also show the temperatures in the UI.

Slipery, 2019-04-01:
I can monitor temperatures on the printer displays both current and target.  Have decided the problem is not with Cura but Marlin.  Something to do with keepalives or busy signals.  When I send M190 s80 for the bed the printer responds with a string of "busy processing" messages and misses the M104 s200 for the extruder command.  Something weird going on as it now wot work from the SD card either.

Slipery, 2019-04-02:
Found the source of the problem but not the solution.  The following Gcode prefixed to the generated code
;Generated with Cura_SteamEngine 3.6.0
M140 S80
M105
M190 S80
M104 S200
M105
M109 S200
Problem is with M190 issued prior to M104 means extruder does not start to heat untill the bed is up to temp.  AnywayI can modify this?

Ghostkeeper, 2019-04-02:
Yeah, you can put your own heating commands in the starting g-code. If you use the setting keys, Cura will not put its own temperature commands in. Something like this:

```
M140 S{material_bed_temperature_layer_0}
M104 S{material_print_temperature_layer_0}
M190 S{material_bed_temperature_layer_0}
M109 S{material_print_temperature_layer_0}
```

Slipery, 2019-04-02:
Thanks for your help thus far much appreciated.
It appears that cura still puts in its own lines at the start as seen at line 7 while my code after " ; Ender 3 Custom Start G-code modified by Mal" is put in.  Put his code in on the printer machine settings should it go somewhere else.

;FLAVOR:Marlin
;TIME:2022
;Filament used: 1.88787m
;Layer height: 0.2
;Generated with Cura_SteamEngine 3.6.0
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code modified by Mal
M140  S(material_bed_temperature_layer_0)
M104 S(material_print_temperature_layer_0)
M190 S(material_bed_temperature_layer_0)
M109 S(material_print_temperature_layer_0)
G28 ; Home all axes

Ghostkeeper, 2019-04-03:
You have to use curly brackets `{}`, not round brackets `()`. Otherwise Cura will not recognise it.

koisdigala, 2021-01-17:
Bug 7835 is a duplicate of this one.

closed 2026-09-06 by GregValiant