Ultimaker/Cura #4928: Cura parameter bug

lafcat, 2018-12-03:
You're welcome...

**Application Version**

3.2.27

**Platform**

Win 7 64 bit

**Printer**

TAZ-6 0.3mm extruder, 1.75mm filament

**Steps to Reproduce**

Trying to set extrusion to 100%, Selected ABS (Village Plastics) but am creating all custom profiles.
I don't use any of the materials defined, so I need to create custom profiles for all my materials.
I try to select extrusion to 100%, but it seems locked to 90%.

**Actual Results**

Extrusion volume locked to 90%.
I finally reduced filament diameter from 1.75mm to 1.6mm to cure under-extrusion problem.
This seemed to work, but is an entirely wrong way to go about this.

**Expected results**

Should have allowed 100% extrusion rate, or above.

**Additional Information**

My extruder is modified for 1.75mm filament.

fieldOfView, 2018-12-03:
Thank you, I guess...

There is no Ultimaker Cura 3.2.27, but there is a Cura Lulzbot Edition with that version. Please refer to them with bugs in their version. See https://code.alephobjects.com/project/profile/10/

nallath, 2018-12-04:
As this isn't "our" version of Cura and the problem is in the profiles provided by lulzbot, i'm closing this.

closed 2018-12-04 by nallath