Ultimaker/Cura #4931: [3.6.0] Wrong machine settings for Wanhao Duplicator i3

ungutknut, 2018-12-04:
**Application Version**
3.6.0

**Platform**
Xubuntu 18.04 64bit; Windows 7 64bit;

**Printer**
Wanhao Duplicator i3

**Steps to Reproduce**
Manage Printers; Add; select Wanhao Duplicator i3; check machine settings;

**Actual Results**
The machine settings are wrong; the start-gcode makes the machine stall after reaching z-endstop

**Expected results**
The machine settings should be right

**Additional Information**
Here's a thread describing the issue in detail with screenshots: https://community.ultimaker.com/topic/25606-cura-360-wanhao-duplicator-i3-wrong-machine-settings/?tab=comments#comment-226799

Loading the Prusa i3 settings and changing the Z-height to 180 is a good workaround. Actually the Duplicator also uses the repetier flavour but it seems to make no difference anyway.

ckielstra, 2018-12-04:
Looks like an issue that was already fixed
https://github.com/Ultimaker/Cura/commit/34b76596f19931cd7498c837e105d95f6db1d445#diff-2cad9b3c673fced30c0d516025e6db56

ungutknut, 2018-12-04:
This should at least fix the travelspeed issue. So now the printer would most likely actually start the print job. But there's still some code missing to switch off the heated bed when the print is done. Like "M140 S0".

alekseisasin, 2018-12-04:
If it so "Looks like the wrong printhead settings are also still there.", please make a Pull Request. Otherwise, it will be always there.

ungutknut, 2018-12-04:
Sorry, I don't know how to do that. I'm not really a coder.

alekseisasin, 2018-12-04:
You can find more details here:
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json
Keep in mind it is a "parent", and printers definitions might override these settings.

drudigger, 2019-12-22:
Any updates on this? Is this fixed in current branch? On cura 2.2 these were the settings that worked for 3 years for me. https://i.imgur.com/uwNoVby.png

Ghostkeeper, 2019-12-23:
The Cura developers can't adjust the settings for printers that we don't have, since we're unable to test them without the hardware. We need to have someone that has the Wanhao Duplicator i3 to provide us with settings and test them (on the latest Cura release).

closed 2019-12-23 by Ghostkeeper

drudigger, 2019-12-23:
Is there anyway i can help with this then if it is a persisting issue, i still have my i3. I would love to do whatever i could.

ckielstra, 2019-12-29:
@drudigger Your help will be appreciated a lot by other i3 users. 
What you need to do is to modify the [printer definition file](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/wanhao_i3.def.json) and the [extruder definition file](https://github.com/Ultimaker/Cura/blob/master/resources/extruders/wanhao_i3_extruder_0.def.json) to be correct for this printer. 

From a quick look it appears the current printer definition for the i3 lacks the settings for a heated bed. 
A good starting point is by comparing the i3 files to the files of a Prusa M3 where the main difference is the Z-height (200 for Prusa, 180mm for i3).

From your screenshot above I see you have the old error where the travel speed is referred to as {travel_speed} but the correct naming is {speed_travel}.

When you have a correct set of printer and extruder files then you can submit through a 'pull request'. If you have troubles with that last part, then I can help you with that.

Ghostkeeper, 2019-12-31:
Cura will actually replace `travel_speed` correctly now just to be backwards compatible.