Ultimaker/Cura #4932: Cura linked Profile to Specific Printer

Thisismydigitalself, 2018-12-04:
**Application Version**
3.6 (same in older versions)

**Platform**
Win10/64/ssd

**Printer**
CFFFP

**Steps to Reproduce**
Start cura clean - without any model loaded.

**Actual Results**
Cura links Profile to last loaded printer but it is "Stared" which means Cura remembers the last used modifyed Profile.

**Expected results**
I would like Cura to not link any profile. i want to be the one to manaully select my profile to be loaded as in the file. i believe there is a link between the type of printer and the profile and i don't like this.

**Additional Information**
please break the link between printer and profile or suggest the user which is the default he/she should load for that printer.

[image]

alekseisasin, 2018-12-04:
What will happen if you select "Update profile with current settings/overrides". Then it is not "Starred" anymore.

Thisismydigitalself, 2018-12-04:
I know, but most of the time it's me tweaking my base profile settings for each specific case. not all models created equal.

alekseisasin, 2018-12-04:
The link indeed exists, between printer and qualities, extruders. 
What do you mean "not all models created equal". You specify a different profile for a specific model.

Thisismydigitalself, 2018-12-05:
I have few profiles for each of my printers.
Beforehand I examine the part about to be printed and i chose the profile most suitable for it, then i tweak settings based on my knowledge and experience. these settings are then saved as a project, e.g. *.3mf files. 

I don't want Cura to alter my profiles. I can understand why Cura remembers my "on-the-fly" changes but these are good for a specific model - not to all.

Personally, i don't like this link between my printers and a default profile. 
I want to be able to manually chose my profiles, and change parameters and settings without these becoming my next time profile - this i don't like.

alekseisasin, 2018-12-05:
What you are asking is much more complex than it sounds. It is "constraint" to have default profiles.
What you can do is to create an own printer definition and define own qualities it will be automatically  linked to them:
https://github.com/Ultimaker/Cura/tree/master/resources/definitions -- printer definitions
https://github.com/Ultimaker/Cura/tree/master/resources/quality -- material qualities.

Thisismydigitalself, 2018-12-05:
A profile is a profile,
A project is a project. 

If there are constraints then at least load the profiles as they are saved and locked in a file and not the last modifyed / altered one in memory.

alekseisasin, 2018-12-05:
I do not know what you mean last modified, but if you want to load a profile then in Cura you can export and import profiles separately. Or do you want to extend project loading with extra options?

Thisismydigitalself, 2018-12-05:
Thank you for this conversation.
My wish is so simple:
If I launch Cura clean, and given there are constraints (which again, i don't like) then at least load linked profile as in saved file - without the star which tells me that what is loaded is different and not the profile as in the saved file. Me, the users, should not bother with altered/stared profiles. these days, each time i launch Cura, i need to go and re-select my default printer's profile for Cura to load it from file. why do i need to do that?

Am i missing something here?
Bottom line is: 
- Why Cura loads a "stared" profile to begin with?
- Why do i need to reload the profile if i want to use my Printer's default profile as i saved it in a file?

GregValiant, 2026-09-10:
I'm going to close this.  If it is still a pertinent issue let us know.

closed 2026-09-10 by GregValiant