Ultimaker/Cura #4933: [3.6] Separated first bottom layers

Liger0, 2018-12-04:
**Application Version**
3.6

**Platform**
Windows 10 x64

**Printer**
Cr-10s

**Steps to Reproduce**
Load the Gcode

**Actual Results**
The bottom layers of each bridge have separated lines

**Expected results**
The lines should all be connected

**Additional Information**
[170-210_PLA_Temp_Tower.zip](https://github.com/Ultimaker/Cura/files/2646294/170-210_PLA_Temp_Tower.zip)
[image]

smartavionics, 2018-12-04:
> The lines should all be connected

Not sure what you are saying here but, hey, why don't you try using the experimental bridge settings and see if you get a result more to your liking?

Liger0, 2018-12-05:
> 
> 
> > The lines should all be connected
> 
> Not sure what you are saying here but, hey, why don't you try using the experimental bridge settings and see if you get a result more to your liking?

It seems like the green perimeters were not engaged and so there were those gaps between the yellow lines. I resetted the cura profile and reinserted the same settings and now it's working. It's not the first time I have to reset my profiles because doing so even with the same settings the result is different.

I'll close it for now.

closed 2018-12-05 by Liger0