Ultimaker/Cura #4934: Mac Os - unable to change linked values ( with chain icon )

marhub, 2018-12-05:
**Application Version**
cura 3.6.0

**Platform**
Mac Os mojave 10.14.1

**Printer**
custom printer

**Steps to Reproduce**
try to edit any value with chain icon, for example **initial layer height**

**Actual Results**
value is not changed

**Expected results**
value should be changed, as in displayed hint - for all prints for this extruder, since it's shared value

**Additional Information**
https://youtu.be/HDSMc3OvOVA

nallath, 2018-12-05:
Could you save this setup in a project file and post your logs?

marhub, 2018-12-06:
not sure if that's what you are asking for :)
[cura-error-project.3mf.zip](https://github.com/Ultimaker/Cura/files/2653672/cura-error-project.3mf.zip)

Let me know it that's it.

nallath, 2019-01-08:
Yeah, that was what I was looking for, but I don't see anything that would explain why it's not working (or at least, I'm not seeing what I anticipated would be the issue).
Has the error occurred since you reset your configuration?