Ultimaker/Cura #4939: Support blocker doesn't work when there are overhanging elements in higher layers

jvandriel, 2018-12-05:
**Application Version**
Cura 3.6.0

**Platform**
macOSX Mojave version 10.14.1 

**Printer**
Creality CR-10S

**Steps to Reproduce**
- Load a model with 90 degree overhanging elements at the top of the model (eg, a box with it cover flaps turned outwards).
- Place supports blockers at the base of the model
- run prepare and watch layer view - support blocker elements will be ignored in those areas where there are overlapping elements at the top of the model

**Actual Results**
All support was printed as if the support blockers weren't there.

**Expected results**
Support material not to be printed at the base of the model and only start where there are no support blocker elements/faces.

**Additional Information**
- This issue is being discussed on the [ultimaker community](https://community.ultimaker.com/topic/25640-cura-support-blocker-not-working-on-macos/)
- [image]
- project: [production.3mf.zip](https://github.com/Ultimaker/Cura/files/2650852/production.3mf.zip)

fieldOfView, 2018-12-16:
Support Blockers are meant to overlap with surfaces that normally need support. You must place them on those areas. 

In other words: Support Blockers are not subtracted from the generated support. Instead, they are meant to mark areas that would normally get supports.

Poeschl, 2019-01-16:
Similar problem when two holes should be without but vertically between them the support should stay.

The blocker can be seen in the first image as light white boxes, and in the layer view only the top one gets blocked.

[image]
[image]

chrisnicola, 2019-01-21:
So I came here after noticing that Cura seems to add unnecessary supports throughout the model when it needs to support the upper layers. In fact if you have "touching buildplate" enabled it will still do this even though it can't actually add the supports for the layer it needs to support.

Here is an example with "everywhere" on and support overhang set at 75 degrees:

[image]

You can see it continues the support to the buildplate and even inside the model somewhat unnecessarily since those areas don't need to be supported.

This is even more unnecessary if "touching buildplate" is enabled.

[image]

It still has the lower support but the areas it is supporting are far less than the 75 degree limit.

I think this explains some other issues I've had with supports showing up on overhangs that definitely were not as steep as I had set the support setting too.

chrisnicola, 2019-01-21:
Just tested this on 4.0 beta and it appears to be fixed! 🏆

closed 2020-08-17 by Ghostkeeper