Ultimaker/Cura #4944: Z Hop should work at travel speed!

sozkani, 2018-12-06:
**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

pediegoivan, 2019-05-15:
I agree or at least a speed control for hop speed should be added!

smartavionics, 2019-05-15:
It does, my travel speed is 300 mm/S and that's the speed it hops at. 

[image]

You need to turn off "Z Hop Only Over Printed Parts".

Ghostkeeper, 2019-05-15:
The horizontal movement of the Z hop is performed at the travel move. The vertical movement is performed at the "Maximum Z Speed". This is a separate setting. You want this to be a separate setting because the Z axis is typically a lot slower than the other axes to move, and causes ringing if there are too many vibrations on it.

closed 2019-05-15 by Ghostkeeper