Ultimaker/Cura #4947: [3.6.0] Troubles of Zig-zag infill

Generator73, 2018-12-06:
Ultimaker Cura 3.6.0 win7-64
Zig-zag infill does not work correctly.

[image]
[image]

smartavionics, 2018-12-06:
Yes, unfortunately there's a regression that breaks various infills and skin. I expect they will fix it sometime. In the meantime, you could try using one of my builds that doesn't suffer from this problem and gives your some other goodies, too. They can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

Thisismydigitalself, 2018-12-06:
@Generator73 
Just checking, - Did you mirror this part in Cura?
https://github.com/Ultimaker/Cura/issues/4858

Generator73, 2018-12-07:
@Thisismydigitalself 
No. I mirrored it in Solidworks.

smartavionics, 2018-12-07:
Switching the infill pattern to lines rather than zig-zag will probably fix the problem.

Generator73, 2018-12-07:
@smartavionics 
Of course. But I need zig-zag.

smartavionics, 2018-12-07:
In that case you must use a different version of Cura to get around this problem.

Generator73, 2018-12-07:
Or I can use other slicer ;)

smartavionics, 2018-12-07:
> Or I can use other slicer ;)

Of course, you always have that option.

Thisismydigitalself, 2018-12-07:
@Generator73 
Before making a mistake and swithing to a differnet slicer :), care to share your stl?  
I am using 3.6 and i don't have this issue, at least i t seem so.

smartavionics, 2018-12-07:
Hello @Thisismydigitalself , whether the problem occurs or not is model and settings dependent. So changing a line width or infill density or similar can make the bug come and go.

Generator73, 2018-12-07:
@Thisismydigitalself 
The same parameters but added another STL model.
[image]

smartavionics, 2018-12-07:
Yes, but the bottom rh part's infill is still screwy.

smartavionics, 2018-12-07:
> Yes. Programmers should fix it.

Indeed so.

Thisismydigitalself, 2018-12-07:
ZigZag InFill is overrated and for the weak, pun intended.
I vote for Gyroid.

ctmakro, 2019-01-10:
same problem as #4559. ran into multiple times. although this problem is going to be fixed in 4.0, I'd still like to post one more example, just for the record.

[image]

v 3.6

BagelOrb, 2020-11-24:
Linked issue is closed because it's fixed, so I'm closing this one as well.

closed 2020-11-24 by BagelOrb