Ultimaker/Cura #4949: Z-axis not moving up as it should in new Cura 3.6.0

TomasNielsen, 2018-12-06:
**Application Version**
Cura 3.6.0

**Platform**
Windows 10

**Printer**
Anet A8

**Steps to Reproduce**
Print a 26 mm cylinder with threads in them

**Actual Results**
At about 15% the Z-axis on my printer still read 0.2 mm and it had printed many layers at the same height.

[PI3_PluggMatt v1.zip](https://github.com/Ultimaker/Cura/files/2655076/PI3_PluggMatt.v1.zip)

**Expected results**
A 15 mm cylinder with threads in the inside.

**Additional Information**
Printer was working fine for 4 month then I upgraded to latest Cura.

smartavionics, 2018-12-06:
Hi @TomasNielsen , I think someone else had a similar problem recently and, if my memory is right, the solution was to disable the accel and/or jerk control in the Cura profile. Hope this helps!

TomasNielsen, 2018-12-07:
HI @smartavionics !
Thanks for the suggestion, I tested it this morning. Disabled both accel & jerk. But after 72% it was still at 0.2 mm. So that did not help in this case.

smartavionics, 2018-12-07:
Looking at the gcode, I notice it's doing retractions with z-hop on layer change and it does this little up/down movement, try disabling the retract on layer change and see if that makes any difference.

TomasNielsen, 2018-12-09:
Tested that now. It ran to 100% completion at 0.2 mm. I will try to downgrade Cura.

smartavionics, 2018-12-09:
OK, but, really, if the machine can't cope with a little jiggle up and down then it's got a problem.

TomasNielsen, 2018-12-09:
No there is something wrong with the latest Cura 3.6. I'm now printing the same .STL with Cura 3.4.1 and it was at 1.6 mm and progressing fine when I left the garage. Cura 3.6 never came higher than 0.2 mm and printed everything on top of the previous layers.

smartavionics, 2018-12-09:
So please zip the gcode that works OK and the gcode that doesn't and attach to this issue so I can compare them. Thanks.

TomasNielsen, 2018-12-09:
[Working3.4.1.zip](https://github.com/Ultimaker/Cura/files/2660740/Working3.4.1.zip)
[NotWorking3.6.0.zip](https://github.com/Ultimaker/Cura/files/2660741/NotWorking3.6.0.zip)
Sure! Here they are.

TomasNielsen, 2018-12-09:
Hi!
This worked super good! The print has a much smoother surface and the threads do not have blobs in them that make them hard to use.
So I would say you found the problem! (I'm eager to hear what you found)

And also 3.6.0 does something to my prints that 3.4.1 does not. Or could it be the disabled z-hops or jerk settings that made the print better?

smartavionics, 2018-12-09:
Great, that confirms the bug and it has to be one of the oddest bugs I've seen in a long time. The reason that the printer is not moving above 0.2mm is because the model file name is PluggMått v1.stl. I know that sounds crazy but it's true. The problem occurs because Cura 3.6 inserts into the gcode comments that contain that name like this excerpt from you bad gcode file:
````
;LAYER:1
M140 S85
G1 F2100 E30.77954
M106 S127.5
;MESH:PluggMått v1.stl
G0 F4500 X97.416 Y105.849 Z0.4
;TYPE:WALL-INNER
G1 F2100 E33.77954
G1 F3000 X97.284 Y106.229 E33.79559
````
I'm guessing that the firmware in your printer doesn't like the å character and somehow that confuses the firmware sufficiently so that the next line in the gcode gets treated as a comment. In the example code above the next line contains the move to the layer above (Z0.4)! and so the move never happens and the nozzle stays at 0.2mm for the whole print.

So a simple workaround should be not to use model files that have file names containing anything but ascii characters.

Is this a bug in Cura or a bug in Marlin or both?

TomasNielsen, 2018-12-10:
HaHa! Yes, that's a little crazy.  ;)
Mått means measure in Swedish. (I'm measuring the threads for a replacement plug for where you prime a water pump)
As a swede that has worked with programming for 20+ years I've seen a lot of weird stuff coming from our three extra characters åäö. Some of the big companies I've consulted for use Chinese characters and then it is not even funny any more.  ;)
UTF-8 usually makes everyone happy.

Glad you found it, good job! I would blame Marlin for the bug but escape the characters in the comment just to save everyone from problems.

fieldOfView, 2018-12-10:
> Maybe Cura should not put UTF-8 characters into those comments.

Should Cura filter the model name before sending it to CuraEngine, or should CuraEngine sanitize the input it gets?

smartavionics, 2018-12-10:
I created a Marlin issue so they are aware of the problem:

https://github.com/MarlinFirmware/Marlin/issues/12620

EgorkaDC, 2018-12-12:
Hey. I also have such a problem. But I'm afraid the problem is not in the characters. Throw off the files, can someone help you figure it out. Printer RepRap. Cura 3.5.9 worked fine, 3.6 refuses to work.
[F_sern8st.zip](https://github.com/Ultimaker/Cura/files/2670776/F_sern8st.zip)

F_sern8 prints perfectly. Korona printed on 0.2 mm.

EgorkaDC, 2018-12-12:
Yes, definitely. This is my problem and you solved it. Thank you very much. Could you write how to fix this problem through Repetier?

EgorkaDC, 2018-12-12:
Will it be possible to continue using Cura?

smartavionics, 2018-12-12:
Just don't use STL file names that contain UTF-8 encoded characters. In your file, it contained the name 'Фишка..stl'. If you just use file names with ASCII characters then you can continue to use Cura 3.6.

EgorkaDC, 2018-12-12:
Thanks again for explaining. I hope this topic will help a lot.

paukstelis, 2018-12-19:
> > Maybe Cura should not put UTF-8 characters into those comments.
> 
> Should Cura filter the model name before sending it to CuraEngine, or should CuraEngine sanitize the input it gets?

That all depends on where it is easier to handle. I'd guess the python side (Cura). Since I was responsible for including these mesh comments into Cura, I'll look into fixing. Unfortunately, it will have to wait till after the holidays.

fieldOfView, 2018-12-19:
Easier is not always better. For me it is also easier to fix it in Cura than in CuraEngine, but I am partial to thinking that CuraEngine should sanitize the input it gets. On the other hand, CuraEngine does not seem to do a lot of sanitizing by design, so perhaps Cura is the better place.

ediciuStsaF, 2019-03-09:
For me i got the same problem ... hope it works by renaming the files.... Thanks so long...

Ghostkeeper, 2019-03-11:
I agree that CuraEngine should sanatise its output such that the printers don't crash on it, if it happens for most printers. You could argue that g-code needs to be ASCII only and then CuraEngine is generating incorrect g-code.

Ghostkeeper, 2019-03-11:
It seems that Ultimaker's printers (at least UM2 and UM3) don't exhibit this problem. Maybe we've done a fix for this a while ago, or maybe it's something that broke in Marlin ever since we forked off.

rburema, 2019-03-14:
We've merged a fix for this in master (of the engine), so it'll probably be in 4.1

See here: https://github.com/Ultimaker/CuraEngine/commit/404760252f958d19b3b24f7e0d27717fd0be5bb0

closed 2019-03-14 by rburema