Ultimaker/Cura #4950: [3.6.0] MacOS Docker Error.

roseg0ld, 2018-12-07:
I am trying to build Cura in Docker. I am quite new to using Docker so I apologize if I'm bringing up a stupid issue. Docker builds Cura just fine but when I go to run it I get the following error:
`Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 665, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/srv/cura/Cura/cura/CuraApplication.py", line 48, in <module>
    from cura.API import CuraAPI
  File "/srv/cura/Cura/cura/API/__init__.py", line 9, in <module>
    from cura.API.Account import Account
  File "/srv/cura/Cura/cura/API/Account.py", line 10, in <module>
    from cura.OAuth2.AuthorizationService import AuthorizationService
  File "/srv/cura/Cura/cura/OAuth2/AuthorizationService.py", line 12, in <module>
    from cura.OAuth2.AuthorizationHelpers import AuthorizationHelpers
  File "/srv/cura/Cura/cura/OAuth2/AuthorizationHelpers.py", line 9, in <module>
    import requests
  File "/usr/local/lib/python3.5/site-packages/requests/__init__.py", line 43, in <module>
    import urllib3
ImportError: No module named 'urllib3'

Original exception was:
Traceback (most recent call last):
  File "cura_app.py", line 135, in <module>
    from cura.CuraApplication import CuraApplication
  File "/srv/cura/Cura/cura/CuraApplication.py", line 48, in <module>
    from cura.API import CuraAPI
  File "/srv/cura/Cura/cura/API/__init__.py", line 9, in <module>
    from cura.API.Account import Account
  File "/srv/cura/Cura/cura/API/Account.py", line 10, in <module>
    from cura.OAuth2.AuthorizationService import AuthorizationService
  File "/srv/cura/Cura/cura/OAuth2/AuthorizationService.py", line 12, in <module>
    from cura.OAuth2.AuthorizationHelpers import AuthorizationHelpers
  File "/srv/cura/Cura/cura/OAuth2/AuthorizationHelpers.py", line 9, in <module>
    import requests
  File "/usr/local/lib/python3.5/site-packages/requests/__init__.py", line 43, in <module>
    import urllib3
ImportError: No module named 'urllib3'`

ChrisTerBeke, 2018-12-11:
Are you using the correct Python version in  your Docker image?

roseg0ld, 2018-12-14:
I built cura build environment before trying to build Cura, is the Python version not set when doing that?

LipuFei, 2018-12-14:
To build Cura, it needs to use the build environment. You can add this flag `-DCMAKE_PREFIX_PATH=<where-you-installed-your-build-environment>`. This flag tells CMake to look for libraries and things first in your build environment folder, otherwise it may use e.g. Python from your system. You can also debug this by looking at the outputs. CMake will show which gcc it has found, which python it has found, etc. Look at the output for those information and check if CMake is using the correct libraries/executable/things.

bebuch, 2019-02-01:
Same on Ubuntu 18.04. (It's docker, it should have the same result on every host system.)

Reproduce by:

```
git clone https://github.com/Ultimaker/Cura.git
cd Cura
docker build -t cura .
docker run cura
```

I think the real Issue is probably in [cura-build-environment](https://github.com/Ultimaker/cura-build-environment). I opened [Issue 57 there](https://github.com/Ultimaker/cura-build-environment/issues/57).