Ultimaker/Cura #4952: No USB connect because of missing baudrate

susisstrolch, 2018-12-07:
**Application Version**
Cura 3.6 and older

**Platform**
all platforms (Linux / Windows / Mac)

**Printer**
Velleman Vertex Delta K8800

**Steps to Reproduce**
Connect the printer, select the Monitor tab

**Actual Results**
Printer is not detected

**Expected results**
Printer should be detected

**Additional Information**
The Vertex Delta K8800 communicates with 500.000 baud. The USBPrinter plugin supports only up to 250.000 baud.

Please, add the 500.000 baud to the list of possible baudrates in USBPrinter plugin.

[USBPrinting.diff.zip](https://github.com/Ultimaker/Cura/files/2656737/USBPrinting.diff.zip)

ChrisTerBeke, 2018-12-14:
It's not only a matter of adding the baudrate to the list, but also having the correct drivers to even send 500k baud over the serial line. Since this baudrate is quite uncommon for 3D printers I'm not sure if these drivers are available (or for which platforms). Anyways, since Ultimaker does not support USB printing on UM machines, this would have to be tested and implemented by the community.

susisstrolch, 2018-12-14:
pullrequest against the 3.6 or against master?

ChrisTerBeke, 2018-12-14:
Master please :)

susisstrolch, 2018-12-14:
closing because of pullrequest "CURA-4952 No USB connect because of missing baudrate".

commit 6600cea6 (2018-12-18): Add missing baudrate to USB connection. 

Extend the baudrate list by 500.000bd to also detect Vertex Delta 3D (K8800). Fixes #4952
closed 2018-12-14 by susisstrolch