Ultimaker/Cura #4955: Suggestion: Add Jerk and Acceleration settings to bridge settings.

SquirrelEngineering, 2018-12-08:
[3.6] Cura
Windows 10
Monoprice maker select 
Bridging PETG
Edges are sagging and stretching out. Not sticking at the edges

I was hoping I could adjust the jerk and accel to play with the time the nozzle spends reversing its direction.

Keep up the good work!

Thanks.

Julius971, 2021-02-28:
Id like to have such s feature as well. I am having issues with bridging caused by acceleration and deceleration. The printer wont utilize the low printing speed set by the experimental bridging setting, due to acceleration values. 
I am missing a feature to slow down the printhead to reach the briding speed as soon as the bridge starts.

smartavionics, 2021-02-28:
My Cura variant which you can find at https://github.com/smartavionics/Cura/releases provides settings to control bridge acceleration and jerk for bridge walls and skins.

GregValiant, 2024-11-24:
@SquirrelEngineering I'm cleaning house around here.  It's been nearly 7 hears and this has not been picked up.  Can I go ahead and close this?

SquirrelEngineering, 2024-11-25:
I would say so

closed 2024-11-25 by GregValiant